MangoAutomation/ma-core-public #799: Rollups - Add ALL rollup type to return all rollups

terrypacker, 2016-08-10:
Since computing 1 rollup results in all rollups being computed we might as well allow returning all of them at the same time.

terrypacker, 2016-08-19:
This could end up being a rollup on the statistics point value endpoint.

terrypacker, 2016-11-08:
After adding to RollupEnum:

Review commented out code in the api module:
IdPointValueStatisticsQuantizerJsonCallback

Then implement the ALL case for all other CSV and JSON statistics quantizer callbacks.

jazdw, 2017-03-31:
@terrypacker Whats the status here? I noticed its in the code and showing up on the data point edit rollup drop-down.

terrypacker, 2017-03-31:
I've just fixed up the REST controller to process the ALL request.  It will need to be tested before we officially start using it though.  Responses look like this:

`[
  {
    "timestamp": 1427536800000,
    "rollup": "ALL",
    "voltage": {
      "AVERAGE": 1.2919492142531948,
      "DELTA": 1.5277884708786946,
      "MINIMUM": 0.9223204871294118,
      "MAXIMUM": 1.8823573208027535,
      "ACCUMULATOR": 1.5277884708786946,
      "SUM": 26686.961014916003,
      "FIRST": 0.9935500585369316,
      "LAST": 1.5277884708786946,
      "INTEGRAL": 422459.7033790715,
      "COUNT": 19461
    },
    "hawaii": {
      "FIRST": "http://localhost:8080/imageValue/hst1490667440891_45.jpg",
      "LAST": "http://localhost:8080/imageValue/hst1490994207560_45.jpg",
      "COUNT": 61
    }
  }
]`

terrypacker, 2017-03-31:
Something to consider is that this isn't supported for CSV currently.  It will simply not write anything out.

terrypacker, 2017-03-31:
See commit infiniteautomation/ma-modules-public@7c1d55b805a7a31ea2b0530966e7d13efeec8a54

terrypacker, 2017-12-01:
This is in the v2 Point Value controller and needs to be tested.  Also CSV support is limited.  Definitely need to test with multiple points in the request.

terrypacker, 2018-04-12:
Closing as this is ready to be reviewed for the UI and any bugs will be opened in a new issue when it starts being used in the UI.

closed 2018-04-12 by terrypacker