MangoAutomation/ma-core-public #805: Data Point Listeners - Put into Data Point Runtime

terrypacker, 2016-08-12:
Data point listeners are used to notify of a point's change.  They are kept in the Runtime Manager because listeners can exist before a point is running and has a Runtime.  Every time a point is updated the point's RT must get the listeners from the Runtime and then notify them.  It would be much more efficient if the listeners were also available in the point Runtime and kept in sync there.

Puckfist, 2016-11-04:
This created concurrency concerns, where a listener can be added at the same time as what it listens to, and the listened point can proceed without ever adding the listener to its listeners.

terrypacker, 2016-11-05:
Reverting changes for 2.8.0 and pushing to review for later release.

Puckfist, 2017-08-18:
Closing as very complex and not directly established as anyone's problem

closed 2017-08-18 by Puckfist