MangoAutomation/ma-core-public #809: Publisher Events - Make alarm levels configurable.

terrypacker, 2016-08-17:
Currently they are hard coded in the PublisherRT and this prohibits not logging them.

terrypacker, 2016-10-27:
This has been added for the 2.7.14 Release.  Test:

- Persistent TCP Publisher
- Modbus Publisher
- Twilio Publisher
- Http Publisher
- Pachube Publisher
- Squwk Publisher 
- BacNet Publisher

Puckfist, 2016-10-28:
I notice on the modbus publisher edit page two of the event levels do not have flags displayed next to them for their alarm level.

Puckfist, 2016-10-28:
Did not work for either events I generated for PTCP - connection refused and point disabled still came at level urgent, despite being set to safety.

Puckfist, 2016-10-28:
A lot of these alarm levels are hardcoded, i.e. PublisherRT:148, 158

Puckfist, 2016-10-28:
https://github.com/infiniteautomation/ma-modules-private/commit/4b939668fed34e4d1962725589a0af8fa88b85fd

Puckfist, 2016-10-28:
Looks like the Twilio events aren't raised for anything, yet.

Puckfist, 2016-11-03:
Looks good.

closed 2016-11-03 by Puckfist