MangoAutomation/ma-core-public #810: Core Table Indexes - Add indexes for searchable tables

terrypacker, 2016-08-18:
With the new RQL endpoints being able to sort and order by any field in a table it has been seen that the queries are quite slow.  Adding indexes to various fields will help.

Tables to review:

dataPoints
dataSources

jazdw, 2016-08-18:
My database has 100,000 points. I am seeing a query time of around 8 seconds on MySQL for the REST call "/rest/v1/data-points?sort(deviceName,name)&limit(200,0)"

Adding an index for deviceName and name didn't seem to help the REST query speed at all.

In the MySQL console running "SELECT SQL_NO_CACHE * FROM mango.datapoints ORDER BY deviceName, name LIMIT 200;" dropped from 1.125s to 0.015s with the index.

jazdw, 2016-08-30:
When doing RQL LIKE queries StringColumnQueryAppender.class wraps the column name with a LOWER() function which causes the index to not be used, I don't think it should do this as the LIKE comparison is case-insensitive anyway.

jazdw, 2016-08-30:
Also restructuring the query for data points into a subquery greatly increases performance, the following results are with an index on the name column.

** Original query (7.7s) **
SELECT dp.id,dp.data,dp.xid,dp.dataSourceId,dp.name,dp.deviceName,dp.enabled,dp.pointFolderId,dp.loggingType,dp.intervalLoggingPeriodType,dp.intervalLoggingPeriod,dp.intervalLoggingType,dp.tolerance,dp.purgeOverride,dp.purgeType,dp.purgePeriod,dp.defaultCacheSize,dp.discardExtremeValues,dp.engineeringUnits,dp.readPermission,dp.setPermission,dp.templateId,ds.name,ds.xid,ds.dataSourceType,template.name FROM dataPoints AS dp join dataSources ds on ds.id = dp.dataSourceId left outer join templates template on template.id = dp.templateId ORDER BY dp.name ASC LIMIT 10 OFFSET 0;

**Subquery (< 0.01s)**
SELECT dp.id,dp.data,dp.xid,dp.dataSourceId,dp.name,dp.deviceName,dp.enabled,dp.pointFolderId,dp.loggingType,dp.intervalLoggingPeriodType,dp.intervalLoggingPeriod,dp.intervalLoggingType,dp.tolerance,dp.purgeOverride,dp.purgeType,dp.purgePeriod,dp.defaultCacheSize,dp.discardExtremeValues,dp.engineeringUnits,dp.readPermission,dp.setPermission,dp.templateId,ds.name,ds.xid,ds.dataSourceType,template.name FROM (SELECT * FROM dataPoints AS dp ORDER BY dp.name ASC LIMIT 10 OFFSET 0) AS dp join dataSources ds on ds.id = dp.dataSourceId left outer join templates template on template.id = dp.templateId;

terrypacker, 2016-09-27:
When testing this we need to check anywhere that previously used LIKE queries via RQL (Data Point Permissions Tools etc.)  To make sure that being case sensitive is ok.

Puckfist, 2017-03-06:
Many indexes were added and released. This is closed?

terrypacker, 2017-03-08:
I think we can close this, we implemented the sub-queries tested and found they weren't ideal in all scenarios.

closed 2018-09-20 by Puckfist