MangoAutomation/ma-core-public #811: Spaces in Path - Core won't start on linux

terrypacker, 2016-08-18:
In a test today the core would not initialize:

`Exception in thread "main" java.lang.IllegalArgumentException: URI is not hierarchical
	at java.io.File.<init>(File.java:418)
	at com.serotonin.util.properties.ReloadingProperties.findFiles(ReloadingProperties.java:214)
	at com.serotonin.util.properties.ReloadingProperties.checkForReload(ReloadingProperties.java:124)
	at com.serotonin.util.properties.ReloadingProperties.<init>(ReloadingProperties.java:54)
	at com.serotonin.util.properties.ReloadingProperties.<init>(ReloadingProperties.java:47)
	at com.serotonin.m2m2.Main.main(Main.java:92)
`

Puckfist, 2016-10-04:
I was not able to replicate with spaces. I did get a failure with a mango directory named 'm n$&^!' though. I tried both tabs and spaces

terrypacker, 2016-12-16:
That exception is actually from something else.  Closing this as not a bug.

closed 2016-12-16 by terrypacker