MangoAutomation/ma-core-public #818: Excel Reports - Page load issues

terrypacker, 2016-08-19:
Page load doesn't seem to work if there are parameters on the URL (sometimes).

Puckfist, 2016-11-22:
Like #660 ? That was found to be related to the template being gone from the templates directory.

Puckfist, 2017-08-15:
I've seen this sometimes after I clear my cache, especially if there are a few report instances on the page I'm loading. Then the first load won't displace any of the report templates, but if I refresh it will come up.

Puckfist, 2018-05-15:
This is what was in my console: 
Uncaught Error: Load timeout for modules: mango-2.0/api
http://requirejs.org/docs/errors.html#timeout
    at B (require.js?v=1526413837:8)
    at D (require.js?v=1526413837:13)
    at require.js?v=1526413837:14

I am content to close this issue since it is being rewritten into the new UI at some point relatively soon, and it is relieved by refreshing the page.

Puckfist, 2018-09-21:
Fixed by new UI

closed 2018-09-21 by Puckfist