MangoAutomation/ma-core-public #823: MangoNoSQL - Wide Query at Start of Shard Fails

terrypacker, 2016-08-23:
The query fails with NPE due to no data being present in the shard before time 0.  This was seen on the FILE_CHANNEL stream.

Puckfist, 2018-04-17:
Was not able to reproduce and didn't see where it would happen. Retest?

terrypacker, 2018-04-17:
Added a test in the NoSQL module ``` testWideQueryNoData ``` this test currently does not complete.  It is iterating over all shards even though they don't exist.  This doesn't produce the bug mentioned here but it could use some work or a discussion I think.

terrypacker, 2018-04-17:
I have commented out the new test in the NoSQL module but it needs to be reviewed.  It looks like I stumbled onto a bug with the delete/purge of data if there is no data in the series.  The test never returns from the after method.