MangoAutomation/ma-core-public #839: REST Long Query Error Handling

terrypacker, 2016-09-27:
If an error occurs in the connection during the return of a large result-set then the exception logging can place a huge load on Mango.  Implement some way to cancel the query or stop the continuous exception generation when streaming results onto a broken connection.

This will likely be done in the StreamableSqlQuery or at least that is a place to start.

terrypacker, 2016-09-28:
Be sure to test that the connection and result set and being close and there are no resource leaks.

Puckfist, 2016-10-18:
In playing with this, I have noticed that /rest/v1/data-points/list does not respect the limit given it by swagger, and seems unlikely to respect any limit statements. It doesn't seem the watchlist builder does either. Probably related.

terrypacker, 2016-11-05:
Confirmed that /rest/v1/data-points/list?limit(10) does in fact work as expected with 10 results.

terrypacker, 2016-11-05:
Closing as both limit and error handling fixed.

closed 2016-11-05 by terrypacker