MangoAutomation/ma-core-public #843: Changes to SQLStatement for REST RQL

terrypacker, 2016-10-05:
The REST RQL endpoints generate SQL via the SQLStatement class.  The following changes have been made for the next 2.7.14 release:

1.  SubQuery is used in the Data Point Dao and adds the WHERE clause to a sub-query on the Joined tables.  This should be tested on all database types.

2.  Removed Joins from COUNT queries when not using WHERE clause

3.  Replaced COUNT(*) with COUNT(DISTINCT {id}) on tables with PKs.  The only known table without a PK that uses this is the UserComment.  Test on all database types.

Puckfist, 2016-10-11:
If this subquery tactic is more efficient, we can work it into the query for startup as well.

Couple comments: 
1. seems the string concatenation at DataPointDao:150 is totally unnecessary,
List<DataPointVO> dps = query(DATA_POINT_SUB_SELECT_PART_1 + " dp.dataSourceId=? " + DATA_POINT_SUB_SELECT_PART_2, new Object[] { dataSourceId },
Since those are all just String literals. This also seems like the only place the sub_select parts are used, so perhaps they can be combined with the " where dataSourceId=? "

2. The various count queries added to AbstractBaseDao are overspecified, where the AS tablePrefix and using the table prefix in the distinct count seem unnecessary (and probably the smallest of performance impacts).

jazdw, 2016-10-31:
Just noticed this in my console output after playing around with some pages, not sure where it came from exactly but I'll write some more tests to test sorting.

> ERROR 2016-10-31 13:07:29,512 (com.infiniteautomation.mango.db.query.StreamableSqlQuery.query:91) - Unknown column 'ds.xid' in 'order clause' 
com.mysql.jdbc.exceptions.jdbc4.MySQLSyntaxErrorException: Unknown column 'ds.xid' in 'order clause'

jazdw, 2016-10-31:
OK I've added tests to Point.spec.js which cover the above issue, there seems to be 2 issues with sorting

1. Sorting by descending device name then point name returns points in incorrect order
2. Sorting by dataSourceXid or dataSourceName returns HTTP 200 successful response but has 0 items in the returned array and throws the above exception which is logged to console

terrypacker, 2016-11-04:
Potentially check FORCE using indexes on MySQL for startup Queries.

terrypacker, 2016-11-05:
Bumping this to 2.8.x as subqueries are turned off for now.

terrypacker, 2017-03-28:
Closing this as we deemed sub-queries unnecessary and slower.  Code remains in the core and can be enabled at a later date if we want.

closed 2017-03-28 by terrypacker