MangoAutomation/ma-core-public #848: Permissions.hasDataPointReadPermssion Utility methods are different

terrypacker, 2016-10-11:
The logic applied to checking data point permissions is not the same for all methods of the Permissions utility.  Decide on the right checking (see REST data point endpoint) for using pre-split sets.

terrypacker, 2017-03-09:
Consider exploding permissions groups when extracting a User from the Database and doing more performant permissions checking via Collections class.  See DataPointFilter in the mangoAPI module on the logic for data point permissions.

terrypacker, 2017-03-09:
Data points and Data Sources have Set<String> in them to use in the Permissions class.

Puckfist, 2017-12-11:
I only found one method to probably be in error, it's the hadDataPointReadPermission around line 145. I think it should have also been checking set permission, but was not

closed 2017-12-22 by Puckfist