MangoAutomation/ma-core-public #851: Core - Discard Extreme Values doesn't prevent setting values

Puckfist, 2016-10-24:
It was observed here: http://infiniteautomation.com/forum/topic/2504/modbus-point-not-observing-discard-extreme-values-during-point-write that a write could be issued to a device even if the value was outside the extreme values range specified in the logging properties. This results in the value being set to the device, but not read and displayed back to the user upon subsequent reads.

resotek, 2016-10-31:
I think the best long-term solution is to add a separate "discard extreme set values" flag. This way the user can limit the value range when setting points, even if it is not desired to discard read values. 
Point discarding should also work if the point is set by another process such as an event handler or a meta-data source. That is, the discard should apply globally regardless of what is trying to change the point value. The goal is to not allow a SCADA write outside of the preset limits.

Based on the past behavior I believe that users expected the discard values to also be applied to writes. Therefore, when initializing the discard extreme set values boolean field, it should be copied from the point logging properties boolean setting. Although that would change existing behavior, I doubt there are many use cases where the user wants to write a point value that is outside the extreme settings.

It would be nice if when setting the point value, the set dialog displayed the point name and the allowed value range. If the point is discarded, an alert should be shown to the user. Perhaps it would be wise to generate an alarm event  if the discarded point value was attempted to be set by another process,

Puckfist, 2016-11-07:
I reflected some on this, and I realized that it is complicated once by the presence of multipliers (which, by the way, I doubt the module I furnished you with takes into account during the set), and secondly by unit conversions.

resotek, 2016-11-23:
The multipliers and unit conversions also confuse the metadata source values when points are referenced in a script, and the threshold values when configuring events.

resotek, 2016-11-23:
Since this issue is not being implemented in core 2.8, I recommend that in the interim the 2.8 help documentation be modified to at least warn the users that it is possible to write extreme Modbus values and not detect those values in the datasource. This issue could have serious implications since Mango gives no indication to the user that a point has been set to a discarded value that is not being displayed.

As far as I remember, the multiplier becomes a divisor during set. Presumably the additive becomes a subtractor. The user does not get to see the original raw register value anywhere in the mango interface when the additive or multiplier values are used. When that point value is used in metadata functions or to set alarm thresholds, it is not obvious which value is being used, especially if unit conversion is enabled for the point. I've been meaning to create an issue about this for a couple of years.

Puckfist, 2017-03-06:
This feature is implemented in the Mango 3.0 core under its own property, Prevent Extreme Sets, which all data points have.

closed 2017-03-23 by Puckfist