MangoAutomation/ma-core-public #854: MySQL Scheduled Backups

terrypacker, 2016-10-26:
Backups have been added: 1b0cff1f5ba2dfcdf08e8c258e543c1195ee6ca9

New env properties if mysql isn't defined on the command line:
db.mysqldump
db.mysql

This has only been tested in OS X so there is likely bugs to find.

Puckfist, 2016-10-31:
I tested this against a remote MySQL, and it worked!

Issues I had:
1. Data source enabled, then backup, then disable, then restore, and the data source isn't polling

2. the reporting of the errors caused by not having mysqldump.exe or mysql.exe on my command line... it seemed as though there was no reporting. No event when mysqldump couldn't be found, only this stack trace, with the UI reporting the backup was scheduled: 
[missing-mysqldump.txt](https://github.com/infiniteautomation/ma-core-public/files/562436/missing-mysqldump.txt)
It also produced a 0 byte backup.

And then when i tried to restore without mysql on the path it said "Database Restored!" in the UI but printed this stack trace:
[missing-mysql.txt](https://github.com/infiniteautomation/ma-core-public/files/562438/missing-mysql.txt)

Puckfist, 2016-10-31:
I am fixing 2 and 3

closed 2016-11-01 by Puckfist