MangoAutomation/ma-core-public #855: Persistent TCP logging - doesn't use RollingIOLog

Puckfist, 2016-10-26:
The Persistent publisher/data source do not use a RollingIOLog, so setting their log level to a low threshhold will generate lots of messages and fill up a user's disk with logs.

terrypacker, 2016-10-27:
Same for the Modbus Publisher and ModbusProcessImageSendThread.

I'm going to implement a RollingProcessLog in the core so we can update these modules after the next release.

terrypacker, 2016-10-28:
Added Rolling Process Log to core: 84a7ff5d2107175de1761082a26424b344c17b33

Puckfist, 2016-10-30:
In trying this, I set the logSize to 0.0001 MB at level=TRACE with 100 logs to keep. I had a .4 file hanging around in the logs/ directory, but I didn't have a .1, .2, or .3! My console is telling me:

ERROR 2016-10-30 08:56:07,063 (com.serotonin.m2m2.util.log.RollingProcessLog.sizeCheck:72) - java.nio.file.NoSuchFileException: path\Core\logs\processLog.PersistentPublisher-1.log.3 

So, I modified it to .1MB and started it up again. I watched .1 get created, then become .2, but then no .1 appeared and the console was telling me this:

ERROR 2016-10-30 09:00:57,030 (com.serotonin.m2m2.util.log.RollingProcessLog.sizeCheck:72) - java.nio.file.NoSuchFileException: C:\IA\Repos\Instance1\ma-core-public\Core\logs\processLog.PersistentPublisher-1.log.1

I am generating hundreds of log messages per second.

Puckfist, 2016-10-30:
Modified the logging to get the whole trace:

ERROR 2016-10-30 09:09:06,058 (com.serotonin.m2m2.util.log.RollingProcessLog.sizeCheck:72) - C:\IA\Repos\Instance1\ma-core-public\Core\logs\processLog.PersistentPublisher-1.log.2 
java.nio.file.NoSuchFileException: C:\IA\Repos\Instance1\ma-core-public\Core\logs\processLog.PersistentPublisher-1.log.2
	at sun.nio.fs.WindowsException.translateToIOException(Unknown Source)
	at sun.nio.fs.WindowsException.rethrowAsIOException(Unknown Source)
	at sun.nio.fs.WindowsException.rethrowAsIOException(Unknown Source)
	at sun.nio.fs.WindowsFileCopy.move(Unknown Source)
	at sun.nio.fs.WindowsFileSystemProvider.move(Unknown Source)
	at java.nio.file.Files.move(Unknown Source)
	at com.serotonin.m2m2.util.log.RollingProcessLog.sizeCheck(RollingProcessLog.java:59)
	at com.serotonin.m2m2.util.log.ProcessLog.log(ProcessLog.java:233)
	at com.serotonin.m2m2.util.log.ProcessLog.info(ProcessLog.java:163)
	at com.serotonin.m2m2.persistent.pub.SyncHandler$PointSync.run(SyncHandler.java:296)
	at com.serotonin.timer.sync.Synchronizer$TaskWrapper.run(Synchronizer.java:155)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

terrypacker, 2016-10-30:
I see this happening on the Point Sync with multiple threads.  The check size method was NOT synchronized at all so it was being called concurrently.  I moved the checkSize into the sync(out) block in the ProcessLog class.

Puckfist, 2016-10-31:
Works! It was even well behaved when my size was 1e-12 MB, printing 1 message into each file and moving on

closed 2016-10-31 by Puckfist