MangoAutomation/ma-core-public #858: Persistent Data Source Throttle Threshold Setting

terrypacker, 2016-10-28:
Add system settings and help for the threshold.

Puckfist, 2016-10-30:
Looks good. I only checked on the help dialogue and not the functionality. My thought is that the existing deployment is a great test, and it seems to work. My only concern would be choosing a default value, or having this on by default. It would seem to me that this behavior is always preferable, but forcing change may not be. So, a high default value should have most smaller users unaffected (for better or worse), whereas a lower value may actually benefit a user whose heap is less gigantic than this default intends.

Puckfist, 2016-10-31:
I moved the wait time into the system settings as well, and I removed the check against the default value to null out that field in the system settings. My thought is that the running publisher is listening for these updates, so it would cause a failure for someone to set the value back to the default for a running publisher.

closed 2016-11-04 by Puckfist