MangoAutomation/ma-core-public #861: Persistent TCP - New Init Logic

terrypacker, 2016-10-30:
The Publisher's point init logic has been changed to send requests in one thread and receive them in another for performance.  We need to test any failure scenarios such as:

1.  Connection is lost during init
2.  One message is lost during init
3.  Point configuration problem
...

Puckfist, 2016-11-08:
Reverting/staving off this issue to the future. There are likely more significant startup time gains to be made in other techniques, but the time to focus on this performance test is not present in this release cycle, and it has been reverted.

Puckfist, 2018-04-25:
The digester, if such a thing is ever implemented, will have to consider that the data source can modify the path or add tags that won't appear in the publisher's JSON.

terrypacker, 2018-09-27:
I guess you would need to confirm that 1 will always and forever equal "LABEL FOR ONE".  Which I guess it should but again we need validation etc so maybe we are making more work for something that may be better off left untouched.  Also considering compression this probably doesn't matter much.