MangoAutomation/ma-core-public #870: Data Point Listener - pointLogged Event

terrypacker, 2016-11-02:
Adding a 'point logged' method to the listener that is fired if the value is saved to the database.  This will be useful for the Persistent publisher to allow publishing the actual values that are saved to the database so there are no overcounts if the logging type on the publishing system is set to anything other than ALL.

Currently all implementations have been left empty so they do nothing with the call as to avoid having to test the changes beyond just in the DataPointRT's EventNotifyWorkItem.

terrypacker, 2016-11-09:
Tested and working with interval logging on published point and log all values on published point.

Puckfist, 2016-11-10:
I reverted some of those in the last commit: https://github.com/infiniteautomation/ma-core-public/commit/57102ab880cb8a195e9d9adf632d8fba4fac4428

I can't recall why I removed the Ready for Testing label before... perhaps it was a discussion we had elsewhere? It seems ready for testing now, but I agree that the publishing should work. I'm not sure anything else is implementing pointLogged yet... so I can check that and perhaps close the issue.

Puckfist, 2016-11-10:
I have confirmed that it's working where it's implemented, but that it isn't implemented in all that many places. For the sake of this release, I am satisfied. I will change the name of this issue, bump it's milestone.

I think we should use this issue to identify point listeners that we would idealize as using or being able to use these logged events (other than publishers). Perhaps we'll revisit the save to cache method at that point.

Puckfist, 2016-11-10:
One module that may truly benefit from something like this is the data file module, if meta points consumed logged events. It has handling for point values in the future, though.

Puckfist, 2017-03-06:
Perhaps there should be a toggle on Meta points to distinguish updating on context and updating on log?

Puckfist, 2017-04-07:
I decided to add this to Meta, and discovered it didn't work for multicasters while I was at it. Here's the commit adding it to Meta: https://github.com/infiniteautomation/ma-modules-private/commit/6dfb3f997548d0fd9fa2bcf659f6a79d2f10472e

Puckfist, 2017-04-17:
I will consider adding it to PointLinks

Puckfist, 2017-04-19:
I have added to PointLinks

Is there anywhere else that could / should listen for it?

Puckfist, 2017-04-19:
I believe that's all the listeners. Scripting data sources don't do context updates (but there is an issue for it) and event detectors don't need the logged event. So, I'd say it's fully implemented, Meta and Point Links got tested. Closing

closed 2017-04-19 by Puckfist