MangoAutomation/ma-core-public #875: Analog Change Detector - new detector type

Puckfist, 2016-11-02:
There exist classes AnalogChangeDetectorVO / RT / Definition but they are not yet functional, and they are not supported yet in eventDetectors.jsp

Puckfist, 2016-11-13:
This is the feature requested in #112 I believe

Puckfist, 2017-08-18:
I feel like this would be relatively simple. The one that currently exists seems like it's geared toward only checking if the value falls or rises (but not both). I feel it is all the same logic and I'm happy to check both or only 1 in the class.

I imagine the RT will have to track its current min/max. Every time it gets a new value, it cancels the timeout task, prunes the list of existing values, if it pruned the min or the max it rechecks the list for the min/max, checks the new value against the min/max, raises the alert if required, updates min/max, puts the value in the list, reschedules the task. The task empties the list of previous values and doesn't reschedule itself.

Puckfist, 2017-08-23:
Okay, I have implemented this. You will need to uncomment ModuleRegistry 932 to use right now

Puckfist, 2017-08-24:
And now I think it's completed

Puckfist, 2017-08-24:
If one felt like getting fancy one could prune values from the list where time < minimumTime && value > minimumValue && time < maximumTime && value < maximumValue but I'm not sure it's worth that much checking... Instead I set it to optionally get values from logging only, instead of change events only.

terrypacker, 2018-01-04:
I've uncommented the ModuleRegistry and lightly tested it.  Seems to work well, could use more testing but will get that in the beta release.

terrypacker, 2018-01-04:
I've re-commented the ModuleRegistry as it does not seem to work as advertised.

Logging every 5s on a Brownian point that changes .01 I have a change limit of 0.009 and 0 duration.  

I'm getting events at seemingly random intervals ie. 15s, 15s, 1min etc.  I think it has to do with no handling for 0 duration timeouts.

The logic for other timeout detectors is different, perhaps taking a look at those might shed some light on the problem.

Puckfist, 2018-01-30:
Ready for retest. I fixed up the 0 duration handling

terrypacker, 2018-03-15:
Retested, still not quite right.  Only raised the alarm 1x then never again.  Here is the JSON config for the virtual point:

```
{
   "dataPoints":[
      {
         "xid":"DP_6e3a8473-8093-4a48-a50f-9665fbe757f1",
         "name":"AnalogChangeDetection",
         "enabled":true,
         "loggingType":"ON_CHANGE",
         "intervalLoggingPeriodType":"MINUTES",
         "intervalLoggingType":"AVERAGE",
         "purgeType":"YEARS",
         "pointLocator":{
            "dataType":"NUMERIC",
            "changeType":{
               "type":"INCREMENT_ANALOG",
               "change":0.01,
               "max":100.0,
               "min":0.0,
               "roll":true,
               "startValue":"1"
            },
            "settable":false
         },
         "eventDetectors":[
            {
               "type":"ANALOG_CHANGE",
               "sourceType":"DATA_POINT",
               "xid":"ED_2287996a-23f1-4b76-be62-7de13d79ce27",
               "alias":"",
               "alarmLevel":"LIFE_SAFETY",
               "durationType":"SECONDS",
               "duration":0,
               "valueEventType":"LOGGED_ONLY",
               "checkDecrease":true,
               "checkIncrease":true,
               "limit":0.009
            }
         ],
         "plotType":"SPLINE",
         "rollup":"NONE",
         "unit":"",
         "chartColour":"",
         "chartRenderer":{
            "type":"IMAGE",
            "timePeriodType":"DAYS",
            "numberOfPeriods":1
         },
         "dataSourceXid":"DS_997094",
         "defaultCacheSize":1,
         "deviceName":"Dashboard Demo",
         "discardExtremeValues":false,
         "discardHighLimit":1.7976931348623157E308,
         "discardLowLimit":-1.7976931348623157E308,
         "intervalLoggingPeriod":1,
         "intervalLoggingSampleWindowSize":0,
         "overrideIntervalLoggingSamples":false,
         "preventSetExtremeValues":false,
         "purgeOverride":false,
         "purgePeriod":1,
         "readPermission":"",
         "setExtremeHighLimit":1.7976931348623157E308,
         "setExtremeLowLimit":-1.7976931348623157E308,
         "setPermission":"",
         "tags":{
         },
         "textRenderer":{
            "type":"ANALOG",
            "useUnitAsSuffix":true,
            "unit":"",
            "renderedUnit":"",
            "format":"0.00"
         },
         "tolerance":0.0
      }
   ]
}
```

terrypacker, 2018-03-15:
Oh also the event message was wrong:
```
Dashboard Demo - AnalogChangeDetection has changed 0.01 or more in 
```

Seems the message was expecting a duration argument but there isn't one on a 0 duration detector.

terrypacker, 2018-03-15:
I see the confusion.  If the event is active then it will not be raised again.  Not sure if that is how we want the 0 duration detection to work.  Should look at how the other detectors use 0 duration.  I was expecting the event to be raised on every poll because the change was > than the limit.

terrypacker, 2018-03-15:
0 Duration working.  But I modified the above configuration to use 10s duration and am not getting any results.  My data source is polling every 5s.

Puckfist, 2018-04-06:
fixed up the messages

Puckfist, 2018-04-11:
ae445c2813a362dc72be1779a8803b280fda3b8e
ec35cc38d410218ea1955f38f9e6a7d612123ded

terrypacker, 2018-04-12:
Closing as complete.

closed 2018-04-12 by terrypacker