MangoAutomation/ma-core-public #880: Data Point Queries - Improve with Join for Event Detectors

terrypacker, 2016-11-02:
There are several data point queries that still use the getRelationalData() method.  These can be improved to make 1 round trip call to the database.

See DataPointDao.DATA_POINT_SELECT_STARTUP for how to use this.

Puckfist, 2017-08-22:
I noticed there were a great deal of opportunities to simply not fetch the relational data. Some, perhaps those in graphical views and watchlists, could still have the relational data excluded, but it was not immediately obvious so I didn't do them right now.

Puckfist, 2017-08-22:
I think there's also a lot of room for gains in the getDataPoint( String xid ) method along these same lines. It unconditionally gets the relational data, but it's quite likely this is not always necessary.

terrypacker, 2017-11-07:
Closing, not going to create a JOIN and try to extract the results.

closed 2017-11-07 by terrypacker