MangoAutomation/ma-core-public #888: Alarm Levels - Revisit Logic and Levels

terrypacker, 2016-11-04:
1.  Add new IGNORE level since none of the existing levels actually totally ignore an alarm.
2.  Review NONE level so that is is logged and handled (and maybe listeners notified, added to user event cache)

Puckfist, 2016-11-11:
Like this issue? https://github.com/infiniteautomation/ma-core-public/issues/892

resotek, 2016-11-23:
For years I've been longing for a level that lies between INFORMATION and URGENT. There are many cases where the operator should receive a warning that is not necessarily urgent, but is more significant than mere information, as it may indicate a problem.  The addition of a level = WARNING would address this, and I believe it would be welcomed by many users. 

Events / Alarm levels don't seem to be described in depth anywhere, but this is how I view alarm levels:
- DO NOT LOG: create an event, but do not spam the event database. Maybe don't send emails
- NONE: create event and add it to event database. Maybe send emails. 
- INFORMATION: no problem, just thought you should know something happened.
- **WARNING**: There may be a problem. You should investigate or be aware, just in case.
- URGENT: There is a problem; you should address it.
- CRITICAL: There is a problem that requires your immediate attention or it may get worse.
- SAFETY: Address this dangerous problem now, before someone is injured.

I've requested a WARNING alarm level tag before, but was told that it was put off until the "event level overhaul" that would allow user-specified levels.I don't think it's been worth waiting 3 years so far for "user specified levels" to get this feature. In fact, I have not noticed a need for custom event levels at all.

snowwindwaverider, 2016-12-05:
I have always just used 3 levels
events - get logged but require no action
alarm - there is a problem with the process the operator should pay attention to or bad things will happen
trip - the process has shutdown and the operator's attention may be required to re-start it when the tripping condition has been cleared.

closed 2017-03-24 by Puckfist