MangoAutomation/ma-core-public #889: DBConvert and NoSQL

Puckfist, 2016-11-04:
Currently, a boot with a database conversion will fail after converting all tables because it cannot cast a PointValueDaoSQL to a NoSQL dao, Stack trace escaping to stderr

Exception in thread "main" java.lang.ExceptionInInitializerError
	at com.infiniteautomation.nosql.MangoNoSqlProxy.shutdown(MangoNoSqlProxy.java:94)
	at com.serotonin.m2m2.db.DatabaseProxy.terminate(DatabaseProxy.java:230)
	at com.serotonin.m2m2.db.DatabaseProxy.initialize(DatabaseProxy.java:137)
	at com.serotonin.m2m2.Lifecycle.databaseInitialize(Lifecycle.java:605)
	at com.serotonin.m2m2.Lifecycle.initialize(Lifecycle.java:283)
	at com.serotonin.m2m2.Main.main(Main.java:115)
Caused by: java.lang.ClassCastException: com.serotonin.m2m2.db.dao.PointValueDaoSQL cannot be cast to com.infiniteautomation.nosql.MangoNoSqlPointValueDao
	at com.infiniteautomation.nosql.MangoNoSqlBatchWriteBehindManager.<init>(MangoNoSqlBatchWriteBehindManager.java:158)
	at com.infiniteautomation.nosql.MangoNoSqlBatchWriteBehindManager.<clinit>(MangoNoSqlBatchWriteBehindManager.java:149)
	... 6 more

terrypacker, 2016-11-05:
Good catch on this one.

Puckfist, 2016-11-07:
Not resolved, getting a lot of these now:

ERROR 2016-11-07 10:07:16,896 (com.serotonin.m2m2.util.timeout.TimeoutTask.run:71) - Uncaught Task Exception 
java.lang.NullPointerException
	at com.infiniteautomation.nosql.MangoNoSqlBatchWriteBehindManager$StatusProvider.scheduleTimeout(MangoNoSqlBatchWriteBehindManager.java:674)
	at com.serotonin.m2m2.util.timeout.TimeoutTask.run(TimeoutTask.java:69)
	at com.serotonin.timer.TimerTask.runTask(TimerTask.java:148)
	at com.serotonin.timer.OrderedTimerTaskWorker.run(OrderedTimerTaskWorker.java:29)
	at com.serotonin.timer.OrderedThreadPoolExecutor$OrderedTask.run(OrderedThreadPoolExecutor.java:278)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(Unknown Source)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(Unknown Source)
	at java.lang.Thread.run(Unknown Source)

Investigation reveals MangoNoSqlBatchWriteBehindManager.proxy == null

closed 2016-11-07 by Puckfist