MangoAutomation/ma-core-public #891: Audit trail - Data point spam or issues?

Puckfist, 2016-11-04:
It seems every time you save a data point it must insert a whole object of its event detectors into the audit table. I can conceivably see why that is, but it may be worth checking into.

More definitely spam, though, is that the dataSourceXid is recorded as null in every save, while the dataSourceId is nowhere to be found.

terrypacker, 2016-11-05:
I cannot replicate the dataSourceXid as null, it is definitely populated when I save it from the legacy data source edit page.  How are you performing the save?  Likely the dataSourceXid member of the data point is not set when your save is occurring.

FYI the audit JSON is generated just like our emport JSON so any non-tracked properties in sero-json will not show up in the JSON.

terrypacker, 2016-11-05:
As for the list of event detectors, we can probably do better on pruning that, will need to investigate after the release.

terrypacker, 2016-12-02:
I've reviewed the code for the event detectors added better detection for changes in the list of detectors.  This will write the entire list to the audit table IFF the list contents change (order not important).

Puckfist, 2017-03-06:
This is still a thing. Using the old data source page, i created a brownian point on my virtual source, saved it, added a change detector, saved it. Saved it. Saved it. Here is my audit trail:

298	DATA_POINT	1	1	2	65	1488824126582	{"dataSourceXid":null}	event.audit.changed|admin (1)|[event.audit.dataPoint|]65|
297	DATA_POINT	1	1	2	65	1488824115323	{"dataSourceXid":null}	event.audit.changed|admin (1)|[event.audit.dataPoint|]65|
296	EVENT_DETECTOR	1	1	1	1	1488824051440	{"xid":"ED_763447","sourceType":"DATA_POINT","alias":"CHANGE","alarmLevel":"NONE","type":"POINT_CHANGE"}	event.audit.added|admin (1)|[event.audit.pointEventDetector|]1|
295	DATA_POINT	1	1	2	65	1488824051436	{"eventDetectors":[{"xid":"ED_763447","sourceType":"DATA_POINT","alias":"CHANGE","alarmLevel":"NONE","type":"POINT_CHANGE"}],"dataSourceXid":null}	event.audit.changed|admin (1)|[event.audit.dataPoint|]65|
294	DATA_POINT	1	1	2	65	1488824032116	{"enabled":true}

Puckfist, 2017-10-05:
At some point this resolved.

closed 2017-10-05 by Puckfist