MangoAutomation/ma-core-public #892: Alarm Level Changes

terrypacker, 2016-11-04:
1.  Add new IGNORE level since none of the existing levels actually totally ignore an alarm.
2.  Review NONE level so that is is logged and handled (and maybe listeners notified, added to user event cache)
3.  Review DO_NOT_LOG level so that it is only handled

terrypacker, 2016-11-04:
Do Not Log Events - Now are only handled 
None Level Events - Log to DB, User Event Cache, Web Sockets, Handled
Ignore - Nothing at all

None Level Events ARE showing up on the HTML 5 Pages (not Dashboards)

terrypacker, 2016-11-05:
Line 108 of EventDao tells us that originally None Level events were ack'ed when inserted.  Is this important to us anymore?

terrypacker, 2016-11-05:
Adding None level events to legacy page toaster.

terrypacker, 2016-11-05:
Audit Events do get Handled, the email could be improved as per #893 after the 2.8.0 release.

terrypacker, 2016-11-05:
Audit events handled in the same way as DO NOT LOG events now.  They only fire event handlers.  The only question here is do we want to fire events for them on the web sockets?  Probably but maybe the event Infrastructure is not the way to do it, also anyone listening to the AuditEventDao could get these messages.

Puckfist, 2016-11-07:
Just a note in addition to that last comment, but Audit events have configurable levels, and that affects a user receiving the audit event emails (in a way that doesn't apply to DO NOT LOG events).

Puckfist, 2016-11-07:
Updating user's receiving of emails automatically to support NONE and DO_NOT_LOG through the addition of IGNORE https://github.com/infiniteautomation/ma-core-public/commit/536b139cae665d5a3a3059be475f615af010371e

closed 2016-11-08 by Puckfist