MangoAutomation/ma-core-public #893: Add JSON Changes to the Audit Event context for use in Emails/Handlers

terrypacker, 2016-11-05:
This would be trivial to implement but would require modifying the email templates or adding the JSON config as an attachment...

Puckfist, 2017-10-12:
I think this is already a thing. I create an event handler on my data point audit event custom email template like:

<#list evt.context?keys as cxt>
${cxt} = ${evt.context[cxt]}
</#list>

and was able to see the properties that I had changed in the email content.

terrypacker, 2017-12-22:
I would suggest adding something to help around this specifically for Audit Events.

terrypacker, 2017-12-22:
I just reviewed the help in that section and it is a little confusing.  Perhaps an example section would be useful.  Plus it implies that some point information is always available, which is only true for point events.

Puckfist, 2018-01-02:
fdc60d4d83646eaa6b74d468ce9fe0ec57291854

closed 2018-01-02 by Puckfist