MangoAutomation/ma-core-public #898: Event Handlers - EventType member change

terrypacker, 2016-11-07:
The Dao has been modified to create the EventType when a handler is retrieved.  This was done to fix the Audit event for delete bug.

Test:

Emport,CRUD operations and Upgrade of DB.

Delete a data point with a detector and check to see it is removed from db.  Also test Emporting points with  detectors.

Puckfist, 2016-11-08:
I have noticed that a SQL error for a duplicate event handler XID-DataPointId breaks the save button for the point and requires the page to be reloaded. This could be handled with a try-catch around DataSourceEditDwr:228:saveDataPoint(dp) but it seems the more consistent way would involve the validation of the detectors, but that isn't occurring in the back end currently.

Puckfist, 2016-11-08:
Fixed: https://github.com/infiniteautomation/ma-core-public/commit/15ffa54be60e06f0861436270268b77487168867

Puckfist, 2016-11-09:
Tested upgrade of old handlers

Puckfist, 2016-11-10:
Works

closed 2016-11-10 by Puckfist