MangoAutomation/ma-core-public #905: PointValueFacade - Improve Performance

terrypacker, 2016-11-11:
Basically rework the facade to use callbacks from the database and expose them in its methods. 

1. Add method getPointValuesBetween with Callback
2. Add method getLatestPointValues with Callback
3. Modify the getPointValuesBetween (initial & final) to use Wide Queries ( #800 #717  )
4. Add method getPointValuesBetween (initial & final) to use Wide Queries with Callback ( #800 #717 )

We also will need to create a MultiPointValueFacade that uses new queries to be added to the PointValueDao:

getLatestPointValues with Multiple IDs. 

Referencing REST Issues: #793 #793 #821

Puckfist, 2016-11-22:
As for the Multi stuff, there is that unused class PVTCollator to either cleanup or use when that time comes.

terrypacker, 2018-03-14:
Closing as worked around in 3.3.x

closed 2018-03-14 by terrypacker