MangoAutomation/ma-core-public #906: module.properties - Dependencies don't work

terrypacker, 2016-11-11:
Fix up the store to accept a module with dependencies in it.  Currently it is rejected when uploaded with an SQL Exception.

The module.properties file supports:

dependencies=<module-name>,<module-name>

jazdw, 2016-11-11:
We should be able to specify dependency versions using something like npm style dependencies - https://docs.npmjs.com/misc/semver

Also the ability to set a module version to a prerelease version such as 1.0.0-beta1 etc would be awesome, so the store can detect that the installed module is a beta and still upgrade it when a 1.0.0 release is available.

We should be able to leverage a library [such as this one](https://github.com/zafarkhaja/jsemver) to make life easier.

Puckfist, 2017-03-07:
The modules page will also ensure you check the download for any dependent modules.

terrypacker, 2017-03-28:
This is done.

jazdw, 2017-05-11:
This is still an issue for me, the Core and Store only check the major and minor version numbers. I have no way to enforce that Dashboard Designer 3.0.2 depends on UI 3.0.2 (3.0.1 will not work!).

We decided to keep the major and minor version numbers of modules the same as the Core so I can't increment the minor version.

I would really like to use the range notations specified in the npm document I linked to above for fine grained dependency control.

Please see 2409e3b3149c66e6954d00d3dabc768f85dd09ca for how we can implement this using the semver library I linked to in Core.

While we are doing this it might be good to consider integrating the versionState into the version instead of having it as a separate field in the module.properties.

Puckfist, 2017-11-07:
Is this considered not ready for testing because it isn't deployed? The store code should have all these features, and thus I would think it ready for testing.

terrypacker, 2017-11-07:
I think my reasoning was that we know there is a problem with the new store code around non-production version dependencies when checking for upgrades so we need to fix that before we can test this?

Puckfist, 2017-11-07:
I still think that's what was requested. It's not a bug. The dependency string is wrong to omit the prerelease version in module.properties files.

Puckfist, 2017-12-11:
I am not doing anything on this as I still consider that the requested functionality.

terrypacker, 2017-12-13:
Test this with the new store, upload a module with a pre-release dependency and try and update it through mango.

terrypacker, 2017-12-22:
Try to download a release version module with a pre-release dependency.

terrypacker, 2017-12-22:
- [ ] Tested installing release version module with with a pre-release dependency.
- [ ] Tested updating release version module with with a pre-release dependency.

- [ ] Tested installing pre-release version module with with a pre-release dependency.
- [ ] Tested updating pre-release version module with with a pre-release dependency.

- [ ] Tested installing pre-release version module with with a release dependency.
- [ ] Tested updating pre-release version module with with a release dependency.

terrypacker, 2017-12-22:
I sort of feel that we should allow people to upgrade missing dependencies.  When I click check for updates I get a message that says I can't upgrade because I have a missing dependency.  At the least we should allow downloading the missing dep.

terrypacker, 2018-01-02:
If I have a released version of ExcelReports 3.3.0 in the store and a locally installed 3.3.0-SNAPSHOT I do not get any option to upgrade in the upgrade request.  This is because the store doesn't use the SNAPSHOT version from the request in checking for the module in the store.

terrypacker, 2018-01-03:
We can't test the new UI for downloading pre-release modules until we fix the problem that the Store wants module names in the form of mangoApi-3.3.0.zip not mangoApi-3.3.0-SNAPSHOT.zip

This has to do with the fact that only one version state of a module can exist in the store so all files are labeled without the pre-release postfix regardless.

Puckfist, 2018-01-03:
Why does that mean we can't use the new UI? It's never permitted the zip file to be named -SNAPSHOT and was not intending to

terrypacker, 2018-01-03:
@Puckfist all that means is that we need to modify the backend or frontend to only request modules without the -SNAPSHOT in the name.  So until we fix that we can't finish the above tests.

terrypacker, 2018-01-04:
This is ready to be close assuming no one can come up with more tests to run...

closed 2018-01-05 by terrypacker