MangoAutomation/ma-core-public #909: ScriptException - Improvements

terrypacker, 2016-11-11:
Improved handling for places where Errors are thrown from scripts and change them to throw ScriptError type exceptions.  Also improve the messages from Java 7 and Java 8 ScriptExceptions see CompiledScriptExector.createScriptError and modify to return ScriptError instead of ScriptException.

1.  Handle potentially un-caught exceptions i.e. NPE
2.  Add utility to better get Line numbers during exception events.

Edits:

**Core**
Compiled Script Executor - Re-throw all exceptions as ScriptError
Compiled Script Executor - Catch all generic Exceptions (DONE for 2.8.0)
Compiled Script Executor - add timing code to execute method (DONE for 2.8.0)

Script Executor - Re-throw all exceptions as ScriptError
Script Executor - Catch all generic Exceptions
Script Executor - Remove prettyScriptMessage() replace with ScriptError.create(e)

**GraphicalViews** 
Script Component

**Meta Data Source**
Historical Meta Point LocatorRT

**Point Links**
PointLinksRT
PointLinksDWR

**Scripting Data Source**

terrypacker, 2016-11-14:
See this reverted commit: cfeea3a3f05f9b9b4e42fe8b290389ebaa033a79 as to some potential code changes.

Puckfist, 2017-04-17:
For 3 I imagine we only need to handle Java 8 style.

Puckfist, 2017-09-18:
I think this could still use some attention. I think script errors may be printing out something like... "line number 41#30 on line 41"

Puckfist, 2017-10-16:
ScriptExecutor.prettyScriptMessage is partially responsible for bad printing. It either needs something like: 

if(message.contains("at line number"))
            message = message.substring(0, message.indexOf("at line number"));

to cut that text off the first message, or it shouldn't even be called anymore since just returning e also renders desirably in simple cases (haven't for the unsimple case yet).

terrypacker, 2019-04-30:
Closing this as it has been reworked in Mango 3.6.  Any additional issues can be opened for more specific problems.

closed 2019-04-30 by terrypacker