MangoAutomation/ma-core-public #913: API returning invalid JSON

jazdw, 2016-11-17:
For a malformed RQL query I am getting a 200 OK response and an invalid JSON response, the API should return 400 Bad Request.

> GET /rest/v1/data-points?eq(pointFolderId,)&sort(deviceName,name)&limit(200) HTTP/1.1
Host: redacted:8080
Connection: keep-alive
Pragma: no-cache
Cache-Control: no-cache
Accept: application/json, text/plain, */*
X-XSRF-TOKEN: redacted
User-Agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/54.0.2840.99 Safari/537.36
Referer: http://redacted:8080/dashboards/watch-list/WL_953943
Accept-Encoding: gzip, deflate, sdch
Accept-Language: en-US,en;q=0.8
Cookie: XSRF-TOKEN=redacted; MANGO8080=redacted

> HTTP/1.1 200 OK
Date: Thu, 17 Nov 2016 17:36:43 GMT
Cache-Control: max-age=0,must-revalidate
messages: 
errors: 
Content-Type: application/json; charset=UTF-8
Content-Encoding: gzip
Vary: Accept-Encoding, User-Agent
ETag: "0e9255693f7acd7a8fe3de956c96ca6c6"
X-Content-Type-Options: nosniff
X-XSS-Protection: 1; mode=block
X-Frame-Options: SAMEORIGIN
Content-Length: 40

> {"items":[],"total"}

jazdw, 2016-11-22:
Further to the above, if I do a query like this 
>/rest/v1/data-points?eq(pointFolderId,dd)&sort(deviceName,name)&limit(200)
It returns results and a total of 99999, even though a folder ID can obviously never be 'dd'.

terrypacker, 2017-01-04:
These have been fixed and are ready for testing.  Basically the StreamableSQLQuery class was catching and logging the exceptions but not bubbling them back up to be returned to the user.

terrypacker, 2017-03-28:
Closing this as fixed for v1, 500 responses are returned for any errors during the query.  We will have to add in additional handling for v2 to translate into 400 errors when v2 is further along.

closed 2017-03-28 by terrypacker