MangoAutomation/ma-core-public #914: Cancellable REST Queries

terrypacker, 2016-11-23:
All non RQL query endpoints should abort their queries if the HTTP Request is cancelled.

Highest priority are the Point Value endpoints (NoSQL/SQL).

Review all other SQL queries

jazdw, 2016-11-23:
You can produce a query which runs in the background even after the HTTP connection is closed by navigating to the watch list page in Dashboards 3.4.0 beta, select a large date range in the date bar and a large number of data points, press the download button and download the point values, hit cancel and you will see that the Java process is still using 1 full CPU core and doing disk IO.

terrypacker, 2016-11-23:
As discussed I used WireShark to see if there is any indication from the page to the server that the request is aborted and it doesn't appear to be the case on OS X using Chrome.

I tried this little script and put a breakpoint in the rest controller to investigate what was happening and it appears there is still a valid OutputStream being used to write the response, in my case it was a GzipOutputStream as the response was gzipped.

`        <script type="text/javascript">
			
        	function reqListener () {
        	  console.log(this.responseText);
        	}

        	var oReq = new XMLHttpRequest();
        	oReq.addEventListener("load", reqListener);
        	oReq.open("GET", "http://localhost:8080/rest/v1/point-values/voltage/latest-multiple-points-single-array");
        	oReq.setRequestHeader("Accept","application/json");
        	oReq.send();
        	oReq.abort();

        </script>`

terrypacker, 2017-11-28:
We have added hooks for the Point Value Dao methods to allow cancelling queries if we can work out how to hook into the lower layers to see the FIN ACK message from the browser.