MangoAutomation/ma-core-public #921: Persistent Publisher - NPE on openConnection if data points have been deleted

Puckfist, 2016-12-06:
NPE PersistentSendThread:557 
ERROR 2016-03-31 17:02:11,223 (com.serotonin.m2m2.rt.publish.SendThread.run:59) - Send thread PersistentSenderRT.SendThread failed with an exception 
java.lang.NullPointerException
 at com.serotonin.m2m2.persistent.pub.PersistentSendThread.openConnection(PersistentSendThread.java:557)
 at com.serotonin.m2m2.persistent.pub.PersistentSendThread.runImpl(PersistentSendThread.java:179)
 at com.serotonin.m2m2.rt.publish.SendThread.run(SendThread.java:56)

Methinks quick fix is...
if(vo == null) { receipts++; continue; }
between 556 and 557

terrypacker, 2016-12-07:
f2af80e6f15da72da1a6d559c2dc32e4758b6c91 Core: There is also no check when validating a PublisherVO to ensure that the points all exist.  I've added code to remove points that don't exist when validating it.

b11336e5eae7d45430caae7b691976f6bee5f948 PersistentTCP: Code has also been added to ensure when starting the publisher it removes missing points from its list during initialization.

Another option is in: PublisherRT.initialize, We could potentially check the point in this method, but the order in which it is called by the PersistentSender is AFTER it's initialize logic is done.

When this is closed we can put the module in the store for 2.8.x as the core changes prevent the bug in a different way.

Puckfist, 2017-03-23:
You added that check in PublisherVO at some point. The only other piece of this that may be left is #948 I believe.

closed 2017-03-23 by Puckfist