MangoAutomation/ma-core-public #922: Env Can data source - fixable?

Puckfist, 2016-12-06:
A user brought a question up on using the EnvCan data source, so I thought I'd give it a whirl and fix what seemed fixable.

Origin of it not working: Canada changed their API around such that 

1) the URLs are different!
What was:
http://climate.weatheroffice.gc.ca/climateData/bulkdata_e.html?StationID=
became:
http://climate.weather.gc.ca/climate_data/bulk_data_e.html?stationID=

2) The returned message is different! Now instead of having XML as the message body it's a file of XML as the message body! I hacked around this by writing a readResponseBody(HttpResponse body) method that consumes the extra things prior to the xml document, like:

public static String readResponseBody(HttpResponse response, int limit) throws IOException {
        InputStream in = response.getEntity().getContent();
        if (in == null)
            return null;
        
        while(in.available() > 0 && in.read() != 0x3c) {}

        Charset charset = ContentType.getOrDefault(response.getEntity()).getCharset();
        if (charset == null)
            charset = Consts.ISO_8859_1;
        InputStreamReader reader = new InputStreamReader(in, charset);
        StringWriter writer = new StringWriter();
        writer.write(0x3c);

        StreamUtils.transfer(reader, writer, limit);

        return writer.toString();
    }

3) sero-utils has some unfortunate XML parsing code, speficially in XMLUtilsTS, the getTextAs methods should be reviewed, as, 

    public static double getTextAsDouble(Element element, double defaultValue) {
        if (element == null)
            return defaultValue;
        try {
            return Double.parseDouble(getElementText(element, null));
        }
        catch (NumberFormatException e) {
            return defaultValue;
        }
    }

misbehaves for empty elements. It'll getElementText defaulting to null, the pass a null to Double.parseDouble which will throw an NPE instead of a NumberFormatException, and that will cause some problems in the calling code.

I did not get EnvCan to work because I didn't get past that NPE for all the elements in this expedition. I think the right choice is to change sero-utils such that the NPE doesn't occur or is caught and the default value is returned. it seemed like one of the calls 147-157 of EnvCanDataSourceRT threw an NPE for every station I tried, but the XML was returned and parsed after steps 1 and 2.

Puckfist, 2016-12-06:
Also, when I was reading the code I was somewhat baffled by the fact that it will query for data starting in 2008, and how all the nextValueTime stuff seems to work. These are probably questions for after it works, though.

Puckfist, 2016-12-06:
I committed the changes that got this working again. I just duplicated the XMLUtilsTS class into the envcan module such that changing sero-utils was not an issue any longer. Tested, was able to get data from a station ID.

It's still a little funky that this module arbitrary starts all periods of interest in 2008. Some stations seem to have a semblence of data as far back as 1848, and some people care very little about anything before 2012 or 2014 or whatever.

Puckfist, 2016-12-06:
It's released as 1.2.1 so we can probably close this at our leisure, unless we're interested in that 2008 thing

closed 2017-01-03 by Puckfist

PhillipDWeeks, 2017-06-15:
Hi there,  I am a Mango 2.8.8 user and have been trying to use the envir can module to restore and replace data from our outside temperature sensors that went down a few month ago in the Halifax area. I hope we can use this data source as a replacement but I cannot find any information on how to use it. Where should I look for user documentation on this module such as the Station ID list etc.?  I have been to the envir can website any help appreciated.