MangoAutomation/ma-core-public #925: Large H2 Database After Dropping Tables

terrypacker, 2016-12-08:
H2 only compacts itself on shutdown for 200ms.  This may not be enough time for a large database to free up some of its unused space.

From H2 Docs:

Compacting a Database
Empty space in the database file re-used automatically. When closing the database, the database is automatically compacted for up to 200 milliseconds by default. To compact more, use the SQL statement SHUTDOWN COMPACT. However re-creating the database may further reduce the database size because this will re-build the indexes. Here is a sample function to do this:

public static void compact(String dir, String dbName,
        String user, String password) throws Exception {
    String url = "jdbc:h2:" + dir + "/" + dbName;
    String file = "data/test.sql";
    Script.execute(url, user, password, file);
    DeleteDbFiles.execute(dir, dbName, true);
    RunScript.execute(url, user, password, file, null, false);
}
See also the sample application org.h2.samples.Compact. The commands SCRIPT / RUNSCRIPT can be used as well to create a backup of a database and re-build the database from the script.

Puckfist, 2017-06-29:
In one extreme example I saw from the forum the user's database shrunk 1000x by being recreated. I like this idea, as I made him do a backup, shutdown, move database, startup, restore sort of process to achieve that.

terrypacker, 2018-06-21:
I have modified the shutdown process of the H2 proxy in hopes that it helps properly perform compaction.

terrypacker, 2018-07-16:
This is an old problem, there is a bug in the H2 code that causes this and is not yet fixed.  Its around LOB handling and is captured in #1291

closed 2018-07-16 by terrypacker