MangoAutomation/ma-core-public #929: Reports - Persist FTL Templates created by Users

terrypacker, 2016-12-13:
Currently the reports module requires the user to add custom templates to the web/ftl folder of the module.  This folder is clobbered on upgrade so this is not an ideal solution.  To fix this we need to:

1. Modify the emailTemplates module.properties property to accept multiple paths (comma separated)
2. change this property to to be web/ftl,web/user-ftl
3. add persistPaths=web/user-ftl 
4. potentially add an upload button on the reports page to allow uploading new FTL templates... Not sure if Freemarker lets you add templates at runtime...

jazdw, 2018-02-12:
@terrypacker Can we look at fixing this, I just modified a report template for a customer and I don't want to be going back making the same modification when they upgrade their modules a month down the track.

Personally I hate the `persistPaths` thing. I think its a flawed design, we should simplify things by adding a data directory for modules, separate to their `/web/modules` directory, e.g. `/data/modules`. Each module could then call `getDataDirectory()` or `getWebDirectory()`.

Actually while we are on the subject, why are modules in the `/web` directory anyway? Can we move their code to just `/modules`, then have `/web/modules` and `/data/modules/`?

terrypacker, 2018-02-16:
I have made a simple fix by allowing a comma separated set of template dirs.

See the branch (main-3.4-multi-emailTemplates) in ma-core-public, ma-core-private, ma-modules-public

See commits:
19aa02e229b5679e16ef0118057193abfb0d43a7
infiniteautomation/ma-core-private@692e576bc5f94a7b25ee357702f508a93b73a7c2
infiniteautomation/ma-modules-public@0c3683ffbb796854ef108ccb538d2c488ec70d09

The comment about the multiple data directories is a larger design discussion that we can look at for 3.4

Puckfist, 2018-03-02:
This issue affects the excel reports module as well, but differently, since it doesn't use the freemarker configuration to find the template. The reports module and the excel reports module have one directory hardcoded (ReportsDwr has getTemplateList conveying that and ExcelReportsCommon has emailTemplatesDirectory)

My thought - rather than getting explosive with the scope - would be to 1) make the reports module load reports like the excel reports does, explicitly from a location. That's because I expect it would find another similarly named and incorrect FTL file on a preferred location as written, like (web/ftl) but I didn't test that yet. and 2) Both the excel reports and reports modules should also explicitly check overrides/web/modules/whatever/web/ftl before checking the default folder.

This thread prompted me to look at the excel reports side of it: https://forum.infiniteautomation.com/topic/3314/excel-report-e-mail-ftl-template

Puckfist, 2018-03-02:
Potentially an issue in my solution is that not loading the template through a configuration may not process included snippets... But, I have to pass in the loader in the configuration to be able to create the template object, so I would guess it would load from the configured, non-override paths to find included snippets. Didn't test that yet, though. Wouldn't be worse that what was, at least, should it be able to find snippets on the loader's path.

closed 2018-04-06 by terrypacker