MangoAutomation/ma-core-public #935: Users - Locale

terrypacker, 2017-01-19:
A new member has been added to the User object and all database tables.  Ensure the SQL upgrade scripts work as well as saving/retrieving users with this value.

See commit: e5853e62491e7eb937110fd377114ff35b8ccc9b

jazdw, 2017-01-20:
Please check if we should use NOT NULL DEFAULT '' in the .sql files and also modify Upgrade13.java to suit

Puckfist, 2017-03-21:
This field does not exist on the legacy users page, so either there needs to be some handling, that page needs to be inaccessible, or it needs to get added to that page too.

terrypacker, 2017-03-21:
I think we just need to add an input to the legacy page.

Puckfist, 2017-03-22:
I will fix this using a hidden input such that you simply cannot change your locale on the legacy users page. One can generate that list if one truly desires

terrypacker, 2017-03-28:
Closing as fixed.

closed 2017-03-28 by terrypacker