MangoAutomation/ma-core-public #936: Users - Name

terrypacker, 2017-01-19:
A new name member has been added to the User object.  Test the SQL upgrade and saving/retrieving a user.

See commit: d93120744a860eb912118f87ba78716895520f61

jazdw, 2017-01-20:
Please check if we should use NOT NULL DEFAULT '' in the .sql files and also modify Upgrade13.java to suit

jazdw, 2017-01-20:
At the moment the validation requires the name field to be set, this can be removed so the legacy users page will still work but this raises an additional issue - if a user is saved on the legacy user page the name and locale fields will be emptied.

Puckfist, 2017-03-22:
The legacy users page supports the name input well and fine, the locale is hidden such that it is simply not editable on the legacy page. I would think that's preferably to a not null default of ''? Can close?

terrypacker, 2017-03-28:
Closing as done.

closed 2017-03-28 by terrypacker