MangoAutomation/ma-core-public #942: Core 3.0.0 & Site Analytics

terrypacker, 2017-02-01:
Ensure that the new pages have a way to report Site Analytics.

jazdw, 2017-03-28:
No this is not implemented, but should be easy. You are talking about Google Analytics yeah?

terrypacker, 2017-03-28:
Yes pretty much, basically any analytics engine that works by inserted some HTML into the page.

jazdw, 2017-04-03:
Hey @terrypacker I think this will be best implemented in a separate Mango module which has an AngularJSModuleDefinition. All it need to do is

- Insert the analytics script
- Get the analytics tracking code from system settings and set it on the analytics script
- Listen to state change events `$rootScope.$on('$stateChangeSuccess', function (event) {})`
- Call `$window.ga('set', 'page', $location.path())` and `$window.ga('send', 'pageview')`

terrypacker, 2017-04-04:
I'm not sure I agree about it being in a separate module.  The system setting is part of the core so I would like to just add that AngularJSModuleDefinition from the Core. The trick will be to only apply it if the site analytics system setting is not empty.  Any thoughts on that one?

jazdw, 2017-04-05:
It will be a new system setting, we can't just inject a script in the head for tracking our SPA (Single Page Application). So it will be a specific Google Analytics property ID setting.

terrypacker, 2017-05-23:
Google analytics are already implemented. The tracking ID is set in the UI settings.

closed 2017-08-03 by terrypacker