MangoAutomation/ma-core-public #947: Core - NumericValue parseNumeric locale specific formatting

Puckfist, 2017-02-10:
The changes to NumericValue.parseNumeric for 3.0 uses only the locale inferred from the language system setting. Perhaps it should get this locale from the user, if possible?

Also it is possible this change would need affect something like the PlainRenderer who is using Double.toString

terrypacker, 2017-03-21:
Potentially make the TextRenderer a 2 way street that would allow inputting a value in the Text Renderer's format and parsing it into a value.  Also ensure that the PlainRenderer uses the User's local to generate the value.

terrypacker, 2017-05-26:
The main problem here is that the outgoing rendering format is different than the incoming parsing of the value.  This should be reviewed.

terrypacker, 2018-04-05:
Fixed up the parse numeric as I feel it should use the User locale if possible.  Avoiding the other discussion.  Perhaps when done testing close and open a new issue or rename this one.

Puckfist, 2018-04-05:
Couple interesting things here:
1) You must still use a format in your analog renderer for the US locale, so a renderer pattern of #.##0,00 has trouble in the UI
2) The ma-set-point-value on the new data point details page is still having values treated with the US format, so my brownian with a value of 14.123,00 (I'm pretending to be in Germany) is appearing in the set value directive as 14.12, which sets the value to 14,12 if I hit enter

terrypacker, 2018-04-17:
I have added additional methods to the TextRenderer that allow passing in a Locale to the getText(...) methods.  Also modified the NumericValue to allow a Locale to be passed in when parsing a value.

Currently the code executes the original methods and uses Common.getLocale() as the locale.  We will want to go through all references to the non-locale methods and see if we can pull in a User's locale for use in the method.

terrypacker, 2018-04-17:
We will also probably want to go through and modify all the renders to use the Locale's when rendering values.  Specifically replace Double.toString() with NumberFormat using the locale that is now available.