MangoAutomation/ma-core-public #950: Core - help page using language modules error

Puckfist, 2017-02-14:
Trying to change a user's language using the button on the far right of the icon bar then navigating to the /help.htm page fails reliably with a decoding error, and a stack trace like this:

$ERROR 2017-02-14 11:15:34,108 (org.apache.taglibs.log.LoggerTag.doStartTag:97) - java.lang.IllegalStateException: Committed
        at org.eclipse.jetty.servlets.gzip.AbstractCompressedStream.resetBuffer(AbstractCompressedStream.java:75)
        at org.eclipse.jetty.servlets.gzip.CompressedResponseWrapper.resetBuffer(CompressedResponseWrapper.java:276)
        at org.eclipse.jetty.servlets.GzipFilter.doFilter(GzipFilter.java:375)
        at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
        at org.springframework.web.filter.ShallowEtagHeaderFilter.doFilterInternal(ShallowEtagHeaderFilter.java:87)
        at com.serotonin.m2m2.web.filter.MangoShallowEtagHeaderFilter.doFilterInternal(MangoShallowEtagHeaderFilter.java:47)
        at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107)
        at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
        at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:585)
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143)
        at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:577)
        at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:223)
        at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1127)
        at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
        at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
        at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1061)
        at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
        at org.eclipse.jetty.server.handler.ContextHandlerCollection.handle(ContextHandlerCollection.java:215)
        at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
        at org.eclipse.jetty.server.Server.handle(Server.java:499)
        at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:310)
        at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:257)
        at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
        at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
        at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
        at java.lang.Thread.run(Thread.java:745)

Puckfist, 2017-02-14:
Switch language to Nederlands, now /help.htm gives this trace:

org.apache.jasper.JasperException: An exception occurred processing JSP page /WEB-INF/jsp/help.jsp at line 6

3:     @author Matthew Lohbihler
4: --%>
5: <%@ include file="/WEB-INF/jsp/include/tech.jsp" %>
6: <tag:page showHeader="${param.showHeader}" showToolbar="${param.showToolbar}" >  
7:   <div id="help" style="margin:0px 30px 0px 30px">
8:     <c:if test="${sessionUser.firstLogin}"><h3><fmt:message key="dox.welcomeToMango"/>!</h3></c:if>
9:     <%-- TODO needs to resolve according to locale rules --%>

Stacktrace:
	at org.apache.jasper.servlet.JspServletWrapper.handleJspException(JspServletWrapper.java:568)
	at org.apache.jasper.servlet.JspServletWrapper.service(JspServletWrapper.java:455)
	at org.apache.jasper.servlet.JspServlet.serviceJspFile(JspServlet.java:405)
	at org.apache.jasper.servlet.JspServlet.service(JspServlet.java:349)
	at org.eclipse.jetty.jsp.JettyJspServlet.service(JettyJspServlet.java:107)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:808)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:587)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143)
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:595)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:223)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1127)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1061)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.Dispatcher.forward(Dispatcher.java:191)
	at org.eclipse.jetty.server.Dispatcher.forward(Dispatcher.java:72)
	at org.springframework.web.servlet.view.InternalResourceView.renderMergedOutputModel(InternalResourceView.java:168)
	at org.springframework.web.servlet.view.AbstractView.render(AbstractView.java:303)
	at org.springframework.web.servlet.DispatcherServlet.render(DispatcherServlet.java:1244)
	at org.springframework.web.servlet.DispatcherServlet.processDispatchResult(DispatcherServlet.java:1027)
	at org.springframework.web.servlet.DispatcherServlet.doDispatch(DispatcherServlet.java:971)
	at org.springframework.web.servlet.DispatcherServlet.doService(DispatcherServlet.java:893)
	at org.springframework.web.servlet.FrameworkServlet.processRequest(FrameworkServlet.java:970)
	at org.springframework.web.servlet.FrameworkServlet.doGet(FrameworkServlet.java:861)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:687)
	at org.springframework.web.servlet.FrameworkServlet.service(FrameworkServlet.java:846)
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790)
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:808)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1669)
	at com.serotonin.web.filter.CharacterSetFilter.doFilter(CharacterSetFilter.java:37)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
	at org.eclipse.jetty.servlets.UserAgentFilter.doFilter(UserAgentFilter.java:83)
	at org.eclipse.jetty.servlets.GzipFilter.doFilter(GzipFilter.java:364)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
	at org.springframework.web.filter.ShallowEtagHeaderFilter.doFilterInternal(ShallowEtagHeaderFilter.java:87)
	at com.serotonin.m2m2.web.filter.MangoShallowEtagHeaderFilter.doFilterInternal(MangoShallowEtagHeaderFilter.java:47)
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107)
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1652)
	at org.eclipse.jetty.servlet.ServletHandler.doHandle(ServletHandler.java:585)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:143)
	at org.eclipse.jetty.security.SecurityHandler.handle(SecurityHandler.java:577)
	at org.eclipse.jetty.server.session.SessionHandler.doHandle(SessionHandler.java:223)
	at org.eclipse.jetty.server.handler.ContextHandler.doHandle(ContextHandler.java:1127)
	at org.eclipse.jetty.servlet.ServletHandler.doScope(ServletHandler.java:515)
	at org.eclipse.jetty.server.session.SessionHandler.doScope(SessionHandler.java:185)
	at org.eclipse.jetty.server.handler.ContextHandler.doScope(ContextHandler.java:1061)
	at org.eclipse.jetty.server.handler.ScopedHandler.handle(ScopedHandler.java:141)
	at org.eclipse.jetty.server.handler.ContextHandlerCollection.handle(ContextHandlerCollection.java:215)
	at org.eclipse.jetty.server.handler.HandlerWrapper.handle(HandlerWrapper.java:97)
	at org.eclipse.jetty.server.Server.handle(Server.java:499)
	at org.eclipse.jetty.server.HttpChannel.handle(HttpChannel.java:310)
	at org.eclipse.jetty.server.HttpConnection.onFillable(HttpConnection.java:257)
	at org.eclipse.jetty.io.AbstractConnection$2.run(AbstractConnection.java:540)
	at org.eclipse.jetty.util.thread.QueuedThreadPool.runJob(QueuedThreadPool.java:635)
	at org.eclipse.jetty.util.thread.QueuedThreadPool$3.run(QueuedThreadPool.java:555)
[... 8 lines cut ...]
	... 56 more
Caused by: java.io.FileNotFoundException: !/WEB-INF/dox/en/help.html
[... cut ...]
	at org.apache.jasper.runtime.JspRuntimeLibrary.include(JspRuntimeLibrary.java:934)
	at org.apache.jsp.WEB_002dINF.jsp.help_jsp$Helper.invoke0(help_jsp.java:267)
	at org.apache.jsp.WEB_002dINF.jsp.help_jsp$Helper.invoke(help_jsp.java:285)
	at org.apache.jsp.tag.web.page_tag.doTag(page_tag.java:354)
	at org.apache.jsp.WEB_002dINF.jsp.help_jsp._jspx_meth_tag_005fpage_005f0(help_jsp.java:132)
	at org.apache.jsp.WEB_002dINF.jsp.help_jsp._jspService(help_jsp.java:96)
	... 59 more

Deleting the work folder doesn't resolve.

terrypacker, 2017-03-28:
There are a few issues here:

1.  There are no help files for any language other than English.

2.  The help.jsp page has code in it that tries to resolve the help file location to a path within WEB-INF/dox/{locale}/help.html  Which is not going to work.

One solution is to create help files for all the language packs, then modify the JSP to resolve the help file location into the language pack module's web dir.  (Or load the files into some cached location at startup).