MangoAutomation/ma-core-public #962: NPE when saving Watchlist via REST

jazdw, 2017-03-28:
https://gist.github.com/jazdw/1f5abb5d678668d2140f90282edce8d1

Occurs due to some watch list types not having a point array (only static WL have a points array). Not sure when this bug was introduced but it definitely used to work.

terrypacker, 2017-03-28:
Something is going on here, this is the second thing I've seen in a few days that has ended up as a null member in the model that previously was working.  The first was in the User model.  These models and backend REST code has not changed in months.  

Fixing with commit: 71be1cda02652ef1319a4eb9f1ac37c1385dad13

jazdw, 2017-03-28:
The NPE in the User model was caused by me adding that isOldHashAlgorithm() method.

I just looked at this issue and its because of a front end change I made about a week ago but you should definitely be able to save a watch list without any points on it, it was only by pure dumb luck that it worked before.

PS I don't particularly like the fix, if I somehow end up with a null member on my watchlist its still going to cause a NPE.

terrypacker, 2017-03-28:
Ok I agree with your complaint.  I'll also put a NP check in the rest methods.

terrypacker, 2017-03-28:
Adding NP checks infiniteautomation/ma-modules-public@6a015c2732823fd32f470c1d43d5f8c6307f79bb

jazdw, 2017-03-28:
Thanks, tested and works.

closed 2017-03-28 by jazdw