MangoAutomation/ma-core-public #963: Session and XSRF Cookie Enhancements

terrypacker, 2017-03-28:
Currently the session cookie is named with the port number in it so that you can login to separate Mangos on the same domain from the same browser.  The XSRF Cookie isn't setup this way and has a static name.

To change this we will want to configure the CookieCsrfTokenRepository as a bean in the MangoSecurityConfiguration so that it uses the port number in the name of the cookie.  This will require the Angular JS configuration be modified to know the new name of the cookie.

jazdw, 2017-04-06:
Another issue with these cookies is that if you have HTTPS enabled on port 8443 and HTTP enabled on port 8080 the session cookie will always be MANGO8080 even if you access it over HTTPS. i.e. the session cookie name is not dependent on the port being used to access Mango, only on what is set in env.properties web.port=

What this means is that if you have HTTPS enabled, you should only use HTTPS to access Mango. If you try and use HTTP after using HTTPS then the browser will not be able set either the session or XSRF cookie.

Puckfist, 2017-10-03:
I think we could simply add an S to the cookie name for the HTTP vs. HTTPS issue, yes? Looks like this may be in M2M2ContextListener and needs to also be updated in MangoSecurityConfiguration on the logout calls

terrypacker, 2017-12-01:
If everyone is happy with using MANGO{port} as the cookie then we can close this.

MANGO8080
MANGO443

jazdw, 2017-12-21:
Commit aa72261 partially addresses my comment on April 6. It does not however address the XSRF token cookie name.

Puckfist, 2018-03-15:
Related? https://github.com/infiniteautomation/ma-dashboards/issues/139

Puckfist, 2018-08-13:
I don't think this issue can be closed. I just attempted to log into two Mango instances, both running on my local machine. One was HTTPS, one was HTTP. With the HTTPS being logged into first, I get an XSRF token that cannot be transmitted or overwritten with the HTTP localhost address. I was able to log into the HTTP Mango instance using the IPv6 localhost address, though

closed 2018-11-06 by terrypacker