MangoAutomation/ma-core-public #964: Copying data point causes exception

jazdw, 2017-03-29:
I have the demo virtual data source installed and it was enabled and running at the time.
I cloned the binary data point, saved it, unticked the property template, saved it again and then enabled it.
The following exception is the result.

```
ERROR 2017-03-29T13:00:17,953 (com.serotonin.m2m2.web.dwr.util.ExceptionDetectionFilter.doFilter:38) - DWR invocation exception 
com.serotonin.ShouldNeverHappenException: org.apache.jasper.JasperException: javax.el.ELException: Error reading 'booleanValue' on type com.serotonin.m2m2.rt.dataImage.PointValueTime
	at com.serotonin.m2m2.web.dwr.BaseDwr.generateContent(BaseDwr.java:391) ~[classes/:?]
	at com.serotonin.m2m2.web.dwr.BaseDwr.setChange(BaseDwr.java:201) ~[classes/:?]
	at com.serotonin.m2m2.web.dwr.BaseDwr.setChange(BaseDwr.java:189) ~[classes/:?]
	at com.serotonin.m2m2.web.dwr.DataPointDetailsDwr.getPointData(DataPointDetailsDwr.java:60) ~[classes/:?]
	at com.serotonin.m2m2.web.dwr.BaseDwr.doLongPoll(BaseDwr.java:592) ~[classes/:?]
	at sun.reflect.GeneratedMethodAccessor665.invoke(Unknown Source) ~[?:?]
	at sun.reflect.DelegatingMethodAccessorImpl.invoke(DelegatingMethodAccessorImpl.java:43) ~[?:1.8.0_121]
	at java.lang.reflect.Method.invoke(Method.java:498) ~[?:1.8.0_121]
	at org.directwebremoting.impl.ExecuteAjaxFilter.doFilter(ExecuteAjaxFilter.java:34) ~[dwr-2.0.11.jar:?]
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) ~[dwr-2.0.11.jar:?]
	at com.serotonin.m2m2.web.dwr.util.TranslationsFilter.doFilter(TranslationsFilter.java:37) ~[classes/:?]
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) ~[dwr-2.0.11.jar:?]
	at com.serotonin.m2m2.web.dwr.util.ExceptionDetectionFilter.doFilter(ExceptionDetectionFilter.java:26) [classes/:?]
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) [dwr-2.0.11.jar:?]
	at com.serotonin.m2m2.web.dwr.util.DwrPermissionFilter.doFilter(DwrPermissionFilter.java:45) [classes/:?]
	at org.directwebremoting.impl.DefaultRemoter$1.doFilter(DefaultRemoter.java:428) [dwr-2.0.11.jar:?]
	at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:431) [dwr-2.0.11.jar:?]
	at org.directwebremoting.impl.DefaultRemoter.execute(DefaultRemoter.java:283) [dwr-2.0.11.jar:?]
	at org.directwebremoting.servlet.PlainCallHandler.handle(PlainCallHandler.java:52) [dwr-2.0.11.jar:?]
	at org.directwebremoting.servlet.UrlProcessor.handle(UrlProcessor.java:101) [dwr-2.0.11.jar:?]
	at org.directwebremoting.servlet.DwrServlet.doPost(DwrServlet.java:146) [dwr-2.0.11.jar:?]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:707) [javax.servlet-api-3.1.0.jar:3.1.0]
	at javax.servlet.http.HttpServlet.service(HttpServlet.java:790) [javax.servlet-api-3.1.0.jar:3.1.0]
	at org.eclipse.jetty.servlet.ServletHolder.handle(ServletHolder.java:845) [jetty-servlet-9.3.11.v20160721.jar:9.3.11.v20160721]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1689) [jetty-servlet-9.3.11.v20160721.jar:9.3.11.v20160721]
	at org.eclipse.jetty.servlets.GzipFilter.doFilter(GzipFilter.java:51) [jetty-servlets-9.3.11.v20160721.jar:9.3.11.v20160721]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1676) [jetty-servlet-9.3.11.v20160721.jar:9.3.11.v20160721]
	at org.springframework.web.filter.ShallowEtagHeaderFilter.doFilterInternal(ShallowEtagHeaderFilter.java:110) [spring-web-4.3.7.RELEASE.jar:4.3.7.RELEASE]
	at com.serotonin.m2m2.web.filter.MangoShallowEtagHeaderFilter.doFilterInternal(MangoShallowEtagHeaderFilter.java:47) [classes/:?]
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107) [spring-web-4.3.7.RELEASE.jar:4.3.7.RELEASE]
	at org.eclipse.jetty.servlet.ServletHandler$CachedChain.doFilter(ServletHandler.java:1676) [jetty-servlet-9.3.11.v20160721.jar:9.3.11.v20160721]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:317) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.invoke(FilterSecurityInterceptor.java:127) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.access.intercept.FilterSecurityInterceptor.doFilter(FilterSecurityInterceptor.java:91) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at com.serotonin.m2m2.web.mvc.spring.security.PermissionExceptionFilter.doFilter(PermissionExceptionFilter.java:32) [classes/:?]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.access.ExceptionTranslationFilter.doFilter(ExceptionTranslationFilter.java:115) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.session.SessionManagementFilter.doFilter(SessionManagementFilter.java:137) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.authentication.AnonymousAuthenticationFilter.doFilter(AnonymousAuthenticationFilter.java:111) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.servletapi.SecurityContextHolderAwareRequestFilter.doFilter(SecurityContextHolderAwareRequestFilter.java:169) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.savedrequest.RequestCacheAwareFilter.doFilter(RequestCacheAwareFilter.java:63) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.session.ConcurrentSessionFilter.doFilter(ConcurrentSessionFilter.java:134) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.authentication.AbstractAuthenticationProcessingFilter.doFilter(AbstractAuthenticationProcessingFilter.java:200) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.authentication.logout.LogoutFilter.doFilter(LogoutFilter.java:121) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.csrf.CsrfFilter.doFilterInternal(CsrfFilter.java:100) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107) [spring-web-4.3.7.RELEASE.jar:4.3.7.RELEASE]
	at org.springframework.security.web.FilterChainProxy$VirtualFilterChain.doFilter(FilterChainProxy.java:331) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.security.web.header.HeaderWriterFilter.doFilterInternal(HeaderWriterFilter.java:66) [spring-security-web-4.1.1.RELEASE.jar:4.1.1.RELEASE]
	at org.springframework.web.filter.OncePerRequestFilter.doFilter(OncePerRequestFilter.java:107) [spring-web-4.3.7.RELEASE.jar:4.3.7.RELEASE]
[... cut ...]
	at com.serotonin.timer.OrderedTimerTaskWorker.run(OrderedTimerTaskWorker.java:29) [serotonin-timer-2.0.9.jar:?]
	at com.serotonin.timer.OrderedThreadPoolExecutor$OrderedTask.run(OrderedThreadPoolExecutor.java:278) [serotonin-timer-2.0.9.jar:?]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1142) [?:1.8.0_121]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:617) [?:1.8.0_121]
	at java.lang.Thread.run(Thread.java:745) [?:1.8.0_121]
```

Puckfist, 2017-03-30:
Could not reproduce from these steps

terrypacker, 2017-03-30:
Basically the cloned point was assigned a data type of Numeric at some point while the PointLocator was setup to be a AlternateBooleanChange type.  By reviewing the code I can see it is possible for the data type to be out of sync with the Virtual point type as there is no enforcement of data type based on the point locator type in a Virtual data source.

jazdw, 2017-04-03:
We figured out that I had folders in my NoSQL database corresponding to the newly created data point id. Its possible I had created data points with these IDs using a previous MySQL or H2 database. Something to keep an eye on but closing for now.

closed 2017-04-03 by jazdw