MangoAutomation/ma-core-public #967: Need smarter request matching for the startup page

jazdw, 2017-03-30:
On startup any HTTP GET request will return the startup page HTML. We really need smarter request matching like what we setup in Spring recently so that we only return the startup page if its
* A GET request
* A request for text/html
* Not a XHR request

Return a HTTP 503 code for all other requests.

terrypacker, 2017-03-30:
Added enhanced matching to StartupContextHandler but only when the existing logic has decided that it is a PAGE type request.  See these commits:

infiniteautomation/ma-core-private@550d9ea5a077f24743a631f7576a6683d91d6d89

857bcaa981488906ca5ecc6f0a3096697ece6a90

jazdw, 2017-03-31:
@terrypacker This is still causing me problems, as it turns out the X-Requested-With header is not added by the browser automatically but it is added by a lot of popular JS frameworks (not AngularJS though).

I'm going to add a global config to our AngularJS module so it gets added to all requests but I think we should also make the startup page always return 503 instead of 200. The page should still get displayed fine.

jazdw, 2017-03-31:
PS, if you want to see the issue this causes in order to test, open a new UI page other than home, hit refresh, restart mango then hit the home link while its starting.

jazdw, 2017-03-31:
I noticed that StartupContextHandler is still checking DefaultPagesDefinition.getStartupUri(), I thought this didn't work?

I've cleaned up the StartupContextHandler in 55ee96c730276e879861df158b4c172e41812163 and made it so only files below /resources and /images can be requested as resources. I also made it return the 503 as described above.

terrypacker, 2017-03-31:
The startup Uri is something we can use but since it runs prior to the Database layer init it must be a hard coded string or accessed from a settings file.

jazdw, 2017-04-05:
I've tested this and it seems good.

closed 2017-04-05 by jazdw