MangoAutomation/ma-core-public #971: Plot type of 'Bar' is not visible on the data point edit page

jazdw, 2017-03-30:
Its listed in the PlotTypes interface but its not visible on the data point edit page plot type drop down.

terrypacker, 2017-03-30:
I noticed this code while fixing this:

`
		if (vo.pointLocator.dataTypeId == <%=DataTypes.NUMERIC%>) {
			show("unitSection");
		} else {
			$("plotType").disabled = true;
			$set("plotType",<%=DataPointVO.PlotTypes.STEP%>);
		}`

Implying that we only allow setting the Plot Type for Numeric type points.  I imagine we want to change this.  How do you want it to work?

terrypacker, 2017-03-30:
Added BAR to the list with this commit: df1174d607c40a931b14662c77eadd6f7d098046

terrypacker, 2017-03-30:
Note that the legacy charts will default to using LINE for both Spline and Bar.  See ImageChartUtils for more info.  At one point I tried to implement a Spline renderer but it didn't behave nicely for all data sets (weird lines etc.) Also as far as I can tell using a BarRenderer isn't an option on an XY Plot in JFree charts so we are sorta stuck with that behavior.

jazdw, 2017-03-30:
> Implying that we only allow setting the Plot Type for Numeric type points. I imagine we want to change this. How do you want it to work?

@terrypacker Probably a question for @Joelhaggar 
I guess its possible that multistate and binary points could have plot types other than step but its kind of odd, no reason why they can't though.

jazdw, 2017-04-03:
@terrypacker Joel said just to leave it. I'm happy with that.

closed 2017-04-03 by jazdw