MangoAutomation/ma-core-public #972: Strange rollup behavior

jazdw, 2017-03-30:
I just copied a numeric data point, enabled it and then viewed it in data point details. Lo and behold the average rollup showed that it had point values before I had even created the point. Strange.. If you look at it with no rollup it shows no values before it was created.

No rollup, last 15 minutes
```
[{"annotation":null,"value":33.894312774376026,"timestamp":1490916027687},{"annotation":null,"value":33.902091713601,"timestamp":1490916032687},{"annotation":null,"value":33.908999283293426,"timestamp":1490916037687},{"annotation":null,"value":33.90951635065316,"timestamp":1490916042687},{"annotation":"Set from web by user: admin","value":6.5,"timestamp":1490916044339},{"annotation":null,"value":6.494427148211747,"timestamp":1490916047687},{"annotation":null,"value":6.498796519987606,"timestamp":1490916052687},{"annotation":null,"value":6.499871420966322,"timestamp":1490916057687},{"annotation":null,"value":6.509675222462268,"timestamp":1490916062687},{"annotation":null,"value":6.513741683037748,"timestamp":1490916067687},{"annotation":null,"value":6.522807738811067,"timestamp":1490916072687},{"annotation":null,"value":6.520484913262519,"timestamp":1490916077687},{"annotation":null,"value":6.529408949853973,"timestamp":1490916082687},{"annotation":null,"value":6.534790741606011,"timestamp":1490916087687},{"annotation":null,"value":6.534343308267325,"timestamp":1490916092687},{"annotation":null,"value":6.538083308634502,"timestamp":1490916097687},{"annotation":null,"value":6.538749240415772,"timestamp":1490916102687},{"annotation":null,"value":6.541806256951888,"timestamp":1490916107687},{"annotation":null,"value":6.540211531347251,"timestamp":1490916112687},{"annotation":null,"value":6.53065837271935,"timestamp":1490916117687},{"annotation":null,"value":6.527555313552542,"timestamp":1490916122687},{"annotation":null,"value":6.526407198009122,"timestamp":1490916127687},{"annotation":null,"value":6.524418849607118,"timestamp":1490916132687},{"annotation":null,"value":6.517731518304668,"timestamp":1490916137687},{"annotation":null,"value":6.5120280833939805,"timestamp":1490916142687},{"annotation":null,"value":6.504319220644387,"timestamp":1490916147687},{"annotation":null,"value":6.494815187324014,"timestamp":1490916152687},{"annotation":null,"value":6.486003367743311,"timestamp":1490916157687},{"annotation":null,"value":6.488554832216368,"timestamp":1490916162687},{"annotation":null,"value":6.489772866618298,"timestamp":1490916167687},{"annotation":null,"value":6.493288740310554,"timestamp":1490916172687},{"annotation":null,"value":6.484586193454473,"timestamp":1490916177687},{"annotation":null,"value":6.482202341383382,"timestamp":1490916182687},{"annotation":null,"value":6.483150444390308,"timestamp":1490916187687},{"annotation":null,"value":6.475501742514355,"timestamp":1490916192687},{"annotation":null,"value":6.477373510553937,"timestamp":1490916197687},{"annotation":null,"value":6.467455777534394,"timestamp":1490916202687},{"annotation":null,"value":6.473232698733202,"timestamp":1490916207687},{"annotation":null,"value":6.463367300605241,"timestamp":1490916212687},{"annotation":null,"value":6.461883766612248,"timestamp":1490916217687},{"annotation":null,"value":6.465820957740805,"timestamp":1490916222687},{"annotation":null,"value":6.465834375578643,"timestamp":1490916227687},{"annotation":null,"value":6.4651282407855755,"timestamp":1490916232687},{"annotation":null,"value":6.474605376824279,"timestamp":1490916237687},{"annotation":null,"value":6.476565675046918,"timestamp":1490916242687},{"annotation":null,"value":6.483961595314233,"timestamp":1490916247687},{"annotation":null,"value":6.492175733399945,"timestamp":1490916252687},{"annotation":null,"value":6.487512761875437,"timestamp":1490916257687},{"annotation":null,"value":6.489519091736377,"timestamp":1490916262687},{"annotation":null,"value":6.48017347428258,"timestamp":1490916267687},{"annotation":null,"value":6.472853678913911,"timestamp":1490916272687},{"annotation":null,"value":6.477580884943724,"timestamp":1490916277687},{"annotation":null,"value":6.469184223638584,"timestamp":1490916282687},{"annotation":null,"value":6.476729689925582,"timestamp":1490916287687},{"annotation":null,"value":6.468265788408156,"timestamp":1490916292687},{"annotation":null,"value":6.4723362766218875,"timestamp":1490916297687},{"annotation":null,"value":6.475784654537253,"timestamp":1490916302687},{"annotation":null,"value":6.475294779073263,"timestamp":1490916307687},{"annotation":null,"value":6.472449731481013,"timestamp":1490916312687},{"annotation":null,"value":6.462845384307116,"timestamp":1490916317687},{"annotation":null,"value":6.466805512919467,"timestamp":1490916322687},{"annotation":null,"value":6.465281265784433,"timestamp":1490916327687},{"annotation":null,"value":6.460256856312896,"timestamp":1490916332687},{"annotation":null,"value":6.459171133032617,"timestamp":1490916337687},{"annotation":null,"value":6.455842411184866,"timestamp":1490916342687},{"annotation":null,"value":6.4595908731896605,"timestamp":1490916347687},{"annotation":null,"value":6.457588992410072,"timestamp":1490916352687},{"annotation":null,"value":6.460077057748275,"timestamp":1490916357687},{"annotation":null,"value":6.458366174003864,"timestamp":1490916362687},{"annotation":null,"value":6.468174756998915,"timestamp":1490916367687},{"annotation":null,"value":6.4687560072180705,"timestamp":1490916372687},{"annotation":null,"value":6.459940350008367,"timestamp":1490916377687},{"annotation":null,"value":6.453517507527838,"timestamp":1490916382687},{"annotation":null,"value":6.460636394100717,"timestamp":1490916387687},{"annotation":null,"value":6.463566991417863,"timestamp":1490916392687},{"annotation":null,"value":6.470484533592198,"timestamp":1490916397687},{"annotation":null,"value":6.47019030828808,"timestamp":1490916402687},{"annotation":null,"value":6.467198534226562,"timestamp":1490916407687},{"annotation":null,"value":6.465350156449368,"timestamp":1490916412687},{"annotation":null,"value":6.459037999448947,"timestamp":1490916417687},{"annotation":null,"value":6.464215196306866,"timestamp":1490916422687},{"annotation":null,"value":6.455496448791803,"timestamp":1490916427687},{"annotation":null,"value":6.461170943523727,"timestamp":1490916432687},{"annotation":null,"value":6.46793589090338,"timestamp":1490916437687},{"annotation":null,"value":6.464158214375735,"timestamp":1490916442687},{"annotation":null,"value":6.461691564450468,"timestamp":1490916447687},{"annotation":null,"value":6.465445818791735,"timestamp":1490916452687},{"annotation":null,"value":6.467244224119299,"timestamp":1490916457687},{"annotation":null,"value":6.471816301464741,"timestamp":1490916462687},{"annotation":null,"value":6.475557652719761,"timestamp":1490916467687},{"annotation":null,"value":6.480707119972681,"timestamp":1490916472687},{"annotation":null,"value":6.482507148012942,"timestamp":1490916477687},{"annotation":null,"value":6.483914275810499,"timestamp":1490916482687},{"annotation":null,"value":6.48836348052363,"timestamp":1490916487687},{"annotation":null,"value":6.488022517684805,"timestamp":1490916492687},{"annotation":null,"value":6.484981087762125,"timestamp":1490916497687},{"annotation":null,"value":6.493409305259423,"timestamp":1490916502687},{"annotation":null,"value":6.488516222692873,"timestamp":1490916507687},{"annotation":null,"value":6.48148882349692,"timestamp":1490916512687},{"annotation":null,"value":6.488185838557344,"timestamp":1490916517687},{"annotation":null,"value":6.492868522468987,"timestamp":1490916522687},{"annotation":null,"value":6.494205444228158,"timestamp":1490916527687},{"annotation":null,"value":6.490889407606804,"timestamp":1490916532687},{"annotation":null,"value":6.499384548631284,"timestamp":1490916537687},{"annotation":null,"value":6.4965930686879325,"timestamp":1490916542687},{"annotation":null,"value":6.502438212161694,"timestamp":1490916547687},{"annotation":null,"value":6.510576921140444,"timestamp":1490916552687},{"annotation":null,"value":6.506737748773246,"timestamp":1490916557687},{"annotation":null,"value":6.498896830239605,"timestamp":1490916562687},{"annotation":null,"value":6.496555879246491,"timestamp":1490916567687},{"annotation":null,"value":6.50083991037099,"timestamp":1490916572687},{"annotation":null,"value":6.503053782241817,"timestamp":1490916577687},{"annotation":null,"value":6.506848625362339,"timestamp":1490916582687},{"annotation":null,"value":6.511738748084893, [... cut ...] on":null,"value":6.526306831173755,"timestamp":1490916670000},{"annotation":null,"value":6.529448098100733,"timestamp":1490916680000},{"annotation":null,"value":6.5318937632799265,"timestamp":1490916690000},{"annotation":null,"value":6.539228527464275,"timestamp":1490916700000},{"annotation":null,"value":6.546141278903813,"timestamp":1490916710000},{"annotation":null,"value":6.5434202768518395,"timestamp":1490916720000},{"annotation":null,"value":6.544842028905904,"timestamp":1490916730000}]
```

terrypacker, 2017-03-30:
This has been an issue for some time and we need to have a discussion on how the various rollups work.  Basically when you start a rollup calculator you provide a 'start value' for it that gets used for periods with no data.  In this case the 'start value' is being collected as the first value for the point and used to backfill the empty periods.

This isn't correct, but making changes to this should involve us reviewing how it will effect all of the rollup calculators and the type of Rollup being performed.

I'd say we push this off till later and draw up a document that describes exactly what we want to happen for each valid data type and rollup combination which will make fixing this up much easier.

jazdw, 2017-03-31:
When we are doing this we should look at #798

terrypacker, 2017-04-05:
Looks like we have a document on this that we can use here:

[http://help.infiniteautomation.com/support/solutions/articles/14000040201-mango-rollups](http://help.infiniteautomation.com/support/solutions/articles/14000040201-mango-rollups)

terrypacker, 2017-05-26:
Also use #831 (Wide Query Implementation)

closed 2017-12-22 by terrypacker