MangoAutomation/ma-core-public #976: Expose Task Info For Thread Pools and Work Items

terrypacker, 2017-04-05:
This involves pulling in some of the changes from the 2.9.x branch that are not yet part of 'main'.  We will get a detailed view of rejected tasks and task durations etc from all the Thread Pool Executors.  There are already pages written to expose them in the 2.9.x Internal Metrics data source branch.

I suggest that we start with pulling in this commit infiniteautomation/ma-modules-public@ f89038ac90a69aaff574336e6ba7f926e5e04383

Then this one:
infiniteautomation/ma-modules-public@399461c1bd2005813e6f1f61b28465a895624221

Which will show some errors in the DWR Code.  From that point we can work backwards by picking out only the code changes we want to include as there is a lot of stuff that was purposefully not brought forward.

Note that the change to the Thread Pool Executors etc are in sero-timer on the 'development' branch which will need to be brought forward as well.

I would strongly consider merging the Sero-Timer project into the Mango Core at this point also.

terrypacker, 2017-04-07:
I've made some headway on this by creating a branch main-sero-timer that contains the serotonin timer code within the mango /src.  I've also cleaned up all the code in main but all the modules will need modifications.

So far the benefits will be:

1.  A tracking repository for rejected execution of High Priority tasks/Work items that is very efficient.
2. Potential to create separate Queue size settings for all the different types of tasks such as:

- Event EventNotifyWorkItem
- Data Point RT Event Notify Work Item
- MangoPingPongTracker
- Set Point Work Item
- Process Work Item
- Email Work Item?
- More

3. Simpler interface to setup tasks that should be queued, how many at most and if not they can be run in parallel immediately.

Puckfist, 2017-05-30:
This has a problem on the Work Items page. It seems to me that everything is displaying in the high priority  work items regardless of the actual priority.

Puckfist, 2017-05-30:
I should amend that. I saw EventNotifyWorkitem in the right category (medium) but NoSQL backup in the wrong category (High instead of low)

terrypacker, 2017-08-03:
You likely saw the scheduled task that fires off the work item.  That task runs at high priority.  Something to note here is that you can't see tasks that are currently running.  There is no way to get those from the ThreadPoolExecutor.  We should probably put something on that page to indicate that the Work items tab only shows queued tasks or figure out a way to track tasks that are running in a thread in the pool.

terrypacker, 2018-03-15:
Closing as this is exposed via REST and DWR.

closed 2018-03-15 by terrypacker