MangoAutomation/ma-core-public #977: Internal Data Source - Automatically Generated Points Log on Average (Not always ideal)

terrypacker, 2017-04-05:
For example the points that do counts such as Data Sources or Data Points or User Sessions should all probably log on instant as it is weird to see .8 user sessions or 10.2 data points.

jazdw, 2017-04-05:
Yeah I agree, logging on average for these is wrong. Another logging type and rollup type we may want to add is mean. Not sure how this would work for floating point values though, I wonder if we could add a integer numeric data type.

terrypacker, 2017-04-12:
I've set all the Numeric type points to log on Change and not be linked to the default template since it does interval average logging.

terrypacker, 2017-04-12:
I guess the remaining question is if we want interval logging or if just on change will suffice.  I prefer on change since there is less overhead.

closed 2017-04-24 by Puckfist