MangoAutomation/ma-core-public #978: API Import/Export endpoint returns no status until complete

jazdw, 2017-04-05:
There is an issue with the API in that no messages are returned while the import is in progress, i.e. I might be 50% of the way through importing 1000 data points but I get no messages, only a status of 50%. Only when it reaches 100% are the messages returned.

terrypacker, 2017-04-06:
This is fixed now, I was trying to avoid synchronization issues by only returning messages at the end.  Also fixed a bug in the cancel method.

This should be tested fairly thoroughly since there is some synchronization involved in keeping the outgoing messages from being clobbered during the import.  Also you don't need to hit the endpoint for status if you are already receiving the WebSocket message as they have the same info in them.  This will avoid performance degradation due to the lack of waiting for the sync block between the importer or the http response thread.

terrypacker, 2017-04-06:
I've been playing with it and it seems like the page is now not showing all the messages that are returned after my fix.  Every time the WebSocket sends a message it seems the page hits the status URL, both will return the list of **EVERY** message that has happened in the import.  If I put a breakpoint on line 434 of JsonEmportRestController I can slow down the calls and the messages seem to show up in order as they should without any missing.

jazdw, 2017-04-06:
I am not using any websockets at all for this, I wrote the page to just poll the REST end point. There doesn't seem to be any real benefit in using a websocket in this scenario as from memory it just pushed out a message every second anyway.

I'm not sure what the issues are with synchronization on your end, I thought there was only one import/export allowed at any one time or has this changed?

Regarding messages, do you mean that every time I hit the REST end-point I get the full list of messages? That is what I would expect. I wouldn't expect the server to keep track of what messages it has already delivered to me and only deliver new ones.

terrypacker, 2017-04-06:
The issues around synchronization are that many people could be asking for the status while the importer is updating the list of messages.  Or that the importer is updating the list of status messages while just one person is asking for them.

@jazdw 
> Regarding messages, do you mean that every time I hit the REST end-point I get the full list of messages? That is what I would expect. I wouldn't expect the server to keep track of what messages it has already delivered to me and only deliver new ones.

Yes that's what I mean.

closed 2017-05-22 by Puckfist