MangoAutomation/ma-core-public #980: Wrong response when using @Secured("ROLE_ADMIN")

jazdw, 2017-04-05:
Using a @Secured("ROLE_ADMIN") annotation on a controller results in an AccessDeniedException which ends up being caught by MangoErrorHandler.

This results in a HTTP 500 error. It should be caught somewhere by Spring and return a 401 or 403 or do a redirect to the authentication entry point.

terrypacker, 2017-04-06:
I don't like using this annotation because we don't really have an role that is labelled that.  Our Admin is superadmin, there is some special handling in to deal with the "ROLE_ADMIN" but I'd like to get away from it.  Preferably we should use:

@PreAuthorize("isAdmin()")

Either way I'll fix the problem you describe here.

terrypacker, 2017-04-06:
Alright so there are a few things going on here that I'll try to lay out.

First there are legacy page requests, page requests from the new app, and REST requests that can all happen from the Browser.  When this happens the browserUrlMatcher will match the request.

So for the 3 cases:

1.  Legacy page requests that are for access denied pages throw a PermissionsException which is translated into an AccessDeniedException which is then caught by the MangoAccessDeniedHandler.  This has been working fine and will redirect to the Unauthorized page defined by the system settings.

2.  Page Requests from the new /dashboards/* for access denied pages are something I've not tested.

3.  Requests for a REST resource below /rest/* are handled by the MangoSpringExceptionHandler which I have now added a special method to handle the AccessDeniedExceptions as you desire.

I have **NOT** seen the MangoErrorHandler catch any AccessDeniedExceptions yet so you will have to give me the exact URL you are using so I can reproduce that.

jazdw, 2017-04-06:
I think I had the annotation on /login/su but I can't replicate it today, I assume your commit above fixed it.

#2 on your list is something that will never happen. /dashboards is an alias for /modules/dashboards/web/ and that is always public

closed 2017-04-06 by jazdw