MangoAutomation/ma-core-public #985: DWR connection error when using HTTP/2

jazdw, 2017-04-12:
Looks like its possibly Jetty closing the connection due to a idle timeout of 50 seconds somewhere.

```
36061: URL_REQUEST
https://localhost:8443/dwr/call/plaincall/MiscDwr.doLongPoll.dwr
Start Time: 2017-04-12 09:58:05.666

t= 4433 [st=    0] +REQUEST_ALIVE  [dt=50005]
                    --> priority = "MEDIUM"
                    --> url = "https://localhost:8443/dwr/call/plaincall/MiscDwr.doLongPoll.dwr"
t= 4433 [st=    0]   +URL_REQUEST_DELEGATE  [dt=2]
t= 4433 [st=    0]      DELEGATE_INFO  [dt=2]
                        --> delegate_blocked_by = "extension uBlock Origin"
t= 4435 [st=    2]   -URL_REQUEST_DELEGATE
t= 4435 [st=    2]   +URL_REQUEST_START_JOB  [dt=50003]
                      --> load_flags = 33026 (BYPASS_CACHE | MAYBE_USER_GESTURE | VERIFY_EV_CERT)
                      --> method = "POST"
                      --> upload_id = "0"
                      --> url = "https://localhost:8443/dwr/call/plaincall/MiscDwr.doLongPoll.dwr"
t= 4435 [st=    2]     +URL_REQUEST_DELEGATE  [dt=1]
t= 4435 [st=    2]        DELEGATE_INFO  [dt=1]
                          --> delegate_blocked_by = "extension Postman Interceptor"
t= 4436 [st=    3]     -URL_REQUEST_DELEGATE
t= 4436 [st=    3]      HTTP_CACHE_GET_BACKEND  [dt=0]
t= 4436 [st=    3]     +HTTP_STREAM_REQUEST  [dt=0]
t= 4436 [st=    3]        HTTP_STREAM_REQUEST_STARTED_JOB
                          --> source_dependency = 36069 (HTTP_STREAM_JOB)
t= 4436 [st=    3]        HTTP_STREAM_REQUEST_BOUND_TO_JOB
                          --> source_dependency = 36069 (HTTP_STREAM_JOB)
t= 4436 [st=    3]     -HTTP_STREAM_REQUEST
t= 4436 [st=    3]     +UPLOAD_DATA_STREAM_INIT  [dt=0]
t= 4436 [st=    3]        UPLOAD_DATA_STREAM_INIT  [dt=0]
                          --> is_chunked = false
                          --> net_error = 0 (?)
                          --> total_size = 228
t= 4436 [st=    3]     -UPLOAD_DATA_STREAM_INIT
                        --> is_chunked = false
                        --> net_error = 0 (?)
                        --> total_size = 228
t= 4436 [st=    3]     +HTTP_TRANSACTION_SEND_REQUEST  [dt=1]
t= 4436 [st=    3]        HTTP_TRANSACTION_HTTP2_SEND_REQUEST_HEADERS
                          --> :authority: localhost:8443
                              :method: POST
                              :path: /dwr/call/plaincall/MiscDwr.doLongPoll.dwr
                              :scheme: https
                              accept: */*
                              accept-encoding: gzip, deflate, br
                              accept-language: en-US,en;q=0.8
                              cache-control: no-cache
                              content-length: 228
                              content-type: text/plain
                              cookie: [115 bytes were stripped]
                              origin: https://localhost:8443
                              pragma: no-cache
                              referer: https://localhost:8443/dataImport.shtm
                              user-agent: Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTML, like Gecko) Chrome/57.0.2987.133 Safari/537.36
t= 4437 [st=    4]       +UPLOAD_DATA_STREAM_READ  [dt=0]
                          --> current_position = 0
t= 4437 [st=    4]          UPLOAD_DATA_STREAM_READ  [dt=0]
                            --> current_position = 0
t= 4437 [st=    4]       -UPLOAD_DATA_STREAM_READ
t= 4437 [st=    4]        HTTP2_STREAM_UPDATE_SEND_WINDOW
                          --> delta = -228
                          --> stream_id = 483
                          --> window_size = 65307
t= 4437 [st=    4]     -HTTP_TRANSACTION_SEND_REQUEST
t= 4437 [st=    4]     +HTTP_TRANSACTION_READ_HEADERS  [dt=50001]
t=54438 [st=50005]        HTTP2_STREAM_ERROR
                          --> description = "SPDY stream closed with status: 8"
                          --> status = -337
                          --> stream_id = 483
t=54438 [st=50005]     -HTTP_TRANSACTION_READ_HEADERS
                        --> net_error = -337 (ERR_SPDY_PROTOCOL_ERROR)
t=54438 [st=50005]   -URL_REQUEST_START_JOB
                      --> net_error = -337 (ERR_SPDY_PROTOCOL_ERROR)
t=54438 [st=50005]    URL_REQUEST_DELEGATE  [dt=0]
t=54438 [st=50005] -REQUEST_ALIVE
                    --> net_error = -337 (ERR_SPDY_PROTOCOL_ERROR)
```

jazdw, 2017-04-12:
OK so in Lifecycle.java we set the HTTP connector to an idle timeout of 30s and the HTTPS connector to 50s. Over HTTP though the DWR long poll connection is not closed after 1 minute though, DWR closes its long poll after 1 minute and starts a new one. So I think this can be solved by setting the HTTPS timeout higher than 1 minute. Will test and post an update.

jazdw, 2017-04-12:
OK so setting the timeout to 70s on L848 of Lifecycle.java fixes this issue. I guess this could have affected any HTTPS connection not just HTTP/2. I'm not going to make the change because I'm not sure if the timeout was set to 50s for any particular reason.

terrypacker, 2017-05-22:
This is now an env property that defaults to 70s

closed 2017-08-03 by terrypacker