MangoAutomation/ma-core-public #987: Raise System Event for Failed Logins

terrypacker, 2017-04-12:
This will allow using event handlers to key off invalid logins and searching events table for these types of events.

terrypacker, 2017-04-12:
Default level to Ignore

jazdw, 2017-04-13:
This is actually pretty important, we should also lock accounts which have multiple failed attempts (provide a system setting for this and possibly make certain users immune).

terrypacker, 2017-05-23:
Raise the event within the MangoAuthenticationFailureHandler.saveExceptionImpl.

terrypacker, 2018-04-04:
The legacy System Settings page gets this setting automatically but we will need to manually add it to the new UI.  @jazdw

jazdw, 2018-04-05:
Done - https://github.com/infiniteautomation/ma-dashboards/commit/475bc0c7f90bc96dc92a296418deef442ad99aa0

closed 2018-04-06 by terrypacker